gnuradio/gnuradio #6981: qa_msg_pair_to_var is flaky

argilo, 2023-12-02:
Recent CI failures:

* https://github.com/gnuradio/gnuradio/actions/runs/7066088459/job/19237340040
* https://github.com/gnuradio/gnuradio/actions/runs/7064478451/job/19232595995

Failures can be produced locally by running `ctest -R qa_msg_pair_to_var --output-on-failure --repeat until-fail:100` together with `stress -c 8` to generate CPU load.

The failures are due to the very short wait times in the tests added in #6950:

https://github.com/gnuradio/gnuradio/blob/8366c28cc65f074c0581a2fd3886566bd6755d44/gr-blocks/python/blocks/qa_msg_pair_to_var.py#L43

https://github.com/gnuradio/gnuradio/blob/8366c28cc65f074c0581a2fd3886566bd6755d44/gr-blocks/python/blocks/qa_msg_pair_to_var.py#L65

/cc @marcusmueller

marcusmueller, 2023-12-02:
ugh, making the test flaky was not the intention! Thanks for bringing this up.

Something is wrong here. To explain:

`tb.start()` only returns when all blocks' `start()` methods have finished.

The message strobe's `start()` is:

```c++
bool message_strobe_impl::start()
{
    d_finished = false;
    d_thread = gr::thread::thread([this] { run(); });

    return block::start();
}

//…

void message_strobe_impl::run()
{
    while (!d_finished) {
        std::this_thread::sleep_for(
            std::chrono::milliseconds(static_cast<long>(d_period_ms)));
        if (d_finished) {
            return;
        }

        message_port_pub(d_port, d_msg);
    }
}
```
so, the moment `tb.start()` returns, the 10 ms timer that will send the message is already running.

In other words, the OS is not managing to wake up the `message_strobe` block executer thread, and get it to notify (and thus wake up) the python blocks thread that is waiting on messages for `msg_pair_to_var` in 40 ms. As far as I know, GNU Radio should be at least *able* to run with RT prio, also on windows, and on [MacOS](https://developer.apple.com/library/archive/documentation/Darwin/Conceptual/KernelProgramming/scheduler/scheduler.html#//apple_ref/doc/uid/TP30000905-CH211-BABCHEEB).

That should effectively prevent the situation where in a flowgraph with two blocks, they're not both woken up within 40 ms! (for comparison, I'd expect an audio system to be woken up every 200 or 400 samples, at 48 kHz, that's less than 9 ms).

So maybe we don't activate rt scheduling on the non-Linux conda CI, @ryanvolz ?

argilo, 2023-12-02:
I'd suggest making the strobe time and sleep time much longer, like in other tests:

https://github.com/gnuradio/gnuradio/blob/2919941fd53bf743bfc2b997f357a670a5d4cf2a/gr-blocks/python/blocks/qa_message.py#L99-L112

marcusmueller, 2023-12-02:
I agree, I should extend the time of that test, it remains a useful test with a longer timeout.

I'm still not convinced that everything is in order when we relatively reliably can trigger situations where a 40ms deadline isn't kept; the problem here might be that `-j N` can't look inside the jobs (which in the case of GR flowgraphs might be heavily multithreaded themselves. If we're running a significant part of the test suite in CPU contention, things get unncessarily slow; maybe adding a `--test-load %CPU_COUNT% ` might be better.

marcusmueller, 2023-12-02:
fix incoming

marcusmueller, 2023-12-02:
> > we relatively reliably can trigger situations where a 40ms deadline isn't kept
> 
> `test_calling` has a 5ms deadline, and it's almost always the culprit. It takes much more load to cause `test_exceptions` to fail.

That was pretty stupid of me, sorry about that!

PR #6983 (2023-12-03): blocks: msq_pair_to_var QA: wait up to 1 s.
Also, fix minor stylistic and include issues

## Description

`test_calling` was effectively only leaving 5 ms for thread wakeup, message delivery and python function calling; that was very flaky in high-load CI environments.

## Related Issue

Fixes #6981

## Which blocks/areas does this affect?

CI

## Testing Done