gnuradio/gnuradio #6994: Frequency Xlating FIR Filter does not generate reproducible results

reald, 2023-12-07:
### What happened?

While analyzing data with gnuradio I stumbled about slightly different results when rerunning my analysis the other day. 

At first I could find out that the **Frequency Xlating FIR Filter** does **produce different results** for the **same input data**. 

There might be more blocks generating non deterministic results.

### System Information

OS: Ubuntu 20.04.5 LTS
GR Installation Method: compile from source
CPU: AMD Ryzen 7 5700G with Radeon Graphics

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

### Steps to reproduce
1.  Generate test data: `dd if=/dev/urandom of=/tmp/test.cfile bs=100000000 count=1`
2.  Run test flow graph
[image]
3.  Compare files
```
md5sum /tmp/file*
860a0ea092969951aee3a93b06559ed1  /tmp/file1.cfile
1973c48093094ec10293e1e2b1ad2268  /tmp/file2.cfile`
```
The begin of both files is identical but then suddenly it begins to differ.

### Expected Behaviour
Same input data must produce same output data.

### Note
GR 3.10.1.1 in Ubuntu 22.04.3 LTS has the same issue.

reald, 2023-12-07:
Flow graph for testing.

[test.zip](https://github.com/gnuradio/gnuradio/files/13603152/test.zip)

willcode, 2023-12-07:
That `dd` and flowgraph produced 2 identical files for me. Linux, Fedora 38, latest GR maint-3.10 branch.

willcode, 2023-12-07:
There were fixes to the rotator (used inside the freq xlating filter) for 3.10.3.0.

reald, 2023-12-07:
Yeah, the test data might a bit short. Rerun it or increase the length. Does not take long to get different results.

willcode, 2023-12-07:
Got a difference after a few tries.

willcode, 2023-12-07:
Ah, bad test. The random number generator is probably producing invalid float values. Try using the random signal source instead, or at least use valid floats for the input. Reopen if the problem is confirmed with good input.

reald, 2023-12-07:
No problem. Here is another option to generate input data. 

[image]

Let it run for a few seconds.

```
md5sum file*
b37ec5c62e267bef210e3a83978cd527  file1.cfile
e648aa019dfb1c9de95900e39eaa1b94  file2.cfile
```

Effect also happended with real measurements...

Flow graph in .zip.

[gen_data.zip](https://github.com/gnuradio/gnuradio/files/13603365/gen_data.zip)

willcode, 2023-12-07:
Reproduced. I added a scale of 32768 to the short-to-float to make sure values were sane. In any case, I can't explain the difference and agree that there shouldn't be any.

Do you see any bad effects, like a dropped sample? Do the differences look like rounding/precision errors, or is the signal totally corrupted when this happens? It's hard to tell using random data.

reald, 2023-12-07:
I stumbled across this after the same input data had a different number of packets with correct CRC. So I placed some file sinks into my flowgraph and the data after the Xlating FIR Filter was not the same after several tries. This was my first catch, there might be more.

I tried building the difference of 2 files and triggering on the magnitude of the difference:
[image]

It showed a delta in magnitude of around 0.5:

[image]

Now I used a signal source instead of a random file:

[image]

This has the disavantage, that when stopping the flowgraph a different number of bytes is written to both files so the checksum method is not possible. But comparing them with `vbindiff` shows following picture:

[image]

It seems that the difference can reach the second byte which is a lot. Surprisingly after a time both signals get the same again... 

[flowgr.zip](https://github.com/gnuradio/gnuradio/files/13603775/flowgr.zip)

willcode, 2023-12-07:
That's really interesting - thanks for diving into it. Don't know what to make of it yet. @argilo @marcusmueller any ideas?

argilo, 2023-12-07:
I would guess it's due to the fact that the thread-per-block scheduler is not deterministic, and so different input chunk sizes can be passed into the rotator, causing it to re-normalize at different points.

argilo, 2023-12-07:
> Is it not happening often enough?

It happens at least once every 512 samples:

https://github.com/gnuradio/volk/blob/d1b1673eb61f936cecee769255203be790b1ad27/kernels/volk/volk_32fc_s32fc_x2_rotator2_32fc.h#L83-L107

argilo, 2023-12-07:
The issue is that `volk_32fc_s32fc_x2_rotator2_32fc` only knows about the samples in the current chunk, and so it normalizes every 512 samples from the first sample in the chunk (plus once more at the end). It has no idea how many samples came before. So you'll get different results in you break the input samples up into different chunk lengths.

willcode, 2023-12-07:
@reald When a different number of packets had the right CRC, was the signal weak (would be on the edge of decoding correctly) or really strong?

argilo, 2023-12-07:
As a hack, it might be possible to force deterministic behaviour by setting the filter block's output multiple to 512.

argilo, 2023-12-07:
> The signal was strong but maybe not suitable for the symbol sync block.

It can be tricky to get Symbol Sync working reliably. Not all timing error detectors are suitable for all signal types, and the parameters may need to be adjusted to achieve quick synchronization if you're dealing with short packets. Another gotcha I sometimes run into is that Symbol Sync doesn't seem to work well if Samples per Symbol is set to a large value (e.g. > 20).

> Where can i set the blocks output?

I don't believe it's possible to do it within GNU Radio Companion. You'd have to edit the generated Python code and add something like `self.<my_freq_xlating_block_name>.set_output_multiple(512)` to the `# Blocks` section.

argilo, 2023-12-07:
Agreed. My motivation for suggesting `set_output_multiple(512)` was to check whether the rotator's phase accumulator normalization was the source of the nondeterministic behaviour.