gnuradio/gnuradio #7006: Flat fader algorithm may have swapped sin and cos

aguahombre, 2023-12-13:
### What happened?

Comparing the original ```flat_fader_impl.cc``` 6 June 2013 and the latest version, it appears that the line
            ```d_phi[n] =  fmod(d_phi[n] + 2 * GR_M_PI * d_fDTs * _GRFASTCOS(alpha_n), 2 * GR_M_PI);```
was originally a SIN function call.
And that the line
           ```float s_q = scale_sin * _GRFASTSIN(d_phi[n]);```
was originally a COS function call.

The change notes don't mention any reason for this change.
Can someone who has access to the original cited paper, check which implementation is correct?

The output of my testing with the latest version of gnuradio fading_model didn't look right

### System Information

OS: ( Windows)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Inspect the ```flat_fader_impl.cc```  github history

willcode, 2023-12-13:
This change was made in 5cf34f170afbc81346c561fb2c1ba178749ab64f about 7 years ago. @osh @marcusmueller @noc0lour maybe you have some insight? It looks like it could be a typo. I don't know the algorithm, but `_GRFASTCOS()` would be more consistent with the rest of the code.

marcusmueller, 2024-02-14:
Yep, there's at least one bug here, but I think it's multiple ones, after scribbling on a few pages of paper:

Citing the used algorithm from A. Alimohammad, S. F. Fard, B. F. Cockburn, and C. Schlegel, “Compact Rayleigh and Rician fading simulator based on random walk processes,” IET Commun., vol. 3, pp. 1333–1342, 2009.

[image]

Vs our code 

```c++
void flat_fader_impl::next_samples(std::vector<gr_complex>& Hvec, int n_samples)
{
    Hvec.resize(n_samples);
    for (int i = 0; i < n_samples; i++) {
        gr_complex H(0, 0);
        for (int n = 1; n < d_N + 1; n++) {
            float alpha_n = (2 * GR_M_PI * n - GR_M_PI + d_theta) / (4 * d_N);
            d_psi[n] =
                fmod(d_psi[n] + 2 * GR_M_PI * d_fDTs * _GRFASTCOS(alpha_n), 2 * GR_M_PI);
            d_phi[n] =
                fmod(d_phi[n] + 2 * GR_M_PI * d_fDTs * _GRFASTCOS(alpha_n), 2 * GR_M_PI);
            float s_i = scale_sin * _GRFASTCOS(d_psi[n]);
            float s_q = scale_sin * _GRFASTSIN(d_phi[n]);
            H += gr_complex(s_i, s_q);
        }

        if (d_LOS) {
            d_psi[0] = fmod(d_psi[0] + 2 * GR_M_PI * d_fDTs * _GRFASTCOS(d_theta_los),
                            2 * GR_M_PI);
            float los_i = scale_los * _GRFASTCOS(d_psi[0]);
            float los_q = scale_los * _GRFASTSIN(d_psi[0]);
            H = H * scale_nlos + gr_complex(los_i, los_q);
        }

        update_theta();
        Hvec[i] = H;
    }
}
```

Which calculates, for the inphase fading factor

$$
\begin{align}
c_i[m] &= C_{\text{scale}} \sum\limits_{n=1}^N \left(
\sum\limits_{k=1}^n\left[ 2\pi f_D T_s \cos(\alpha_n[m]) + \psi_k\right]
\right)\\
&=C_{\text{scale}} \sum\limits_{n=1}^N \left(
 2\pi f_D T_s n \cos(\alpha_n[m])
\right)\\
\end{align}$$

Note that there's no $m$ in the $2\pi f_D t_s \cos()$ in our calculation! I think the intention was to solve that by having `H += gr_complex(s_i, s_q); Hvec[i] = H;` alas, trigonometric functions aren't linear, so that doesn't work.

And yes, the `sin` in there is wrong, needs to be `s_i = … cos( … cos(…))` and `s_i = … cos( … sin(…))`, but that just amounts to a different initializaton to `d_phi`, so while that is a deviation from the model, I don't think it hurts.