gnuradio/gnuradio #7007: UDP sink block works weird

Lunny1993, 2023-12-14:
### What happened?

My flowgraph add UDP-SINK to send spectrum data as vector type，but UDP sending always stops abnormally quickly.
This is true for several versions of GNURadio，and I wonder if the UDP-SINK block is faulty or if I'm using it the wrong way？

### System Information

OS: Windows 10
GR Installation Method: radioconda installer

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0 (Python 3.11.6)

### Steps to Reproduce the Problem

1、My gnuradio by radioconda on windows version：
[image]
2、My flowgraph to get and send spectrum data as vector type over UDP-SINK block：
[image]
[image]
3、In fact, after the flowgraph runs, only a few packets of UDP data can be sent, and then the UDP packets stop sending：
[image]
4、I tried lowering the sample rate or lowering the FFT size etc, but it still didn't work.
5、I noticed the "NOTE" for "UDP-SINK", but how to add the following lines to "/etc/sysctl.conf" on the windows version?
[image]

Can anyone fix it? Ask for advice! Thanks a lot!

### Relevant log output

```shell
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_108200; UHD_4.5.0.0-release
[INFO] [B200] Detected Device: B205mini
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 32.000000 MHz... 
[INFO] [B200] Actually got clock rate 32.000000 MHz.
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
```

willcode, 2023-12-14:
What happens if the USRP source (or a throttled signal source) is connected directly to the UDP sink. Try without vectors first, then with stream-to-vector. Also try with smaller packet sizes, e.g., 1000, so jumbo frames are not required.

My guess at the moment is that this is a windows networking or firewall effect of some sort, but that's just a wild guess at this point.

jmfriedt, 2024-11-05:
I believe a simpler way to reproduce is:
[image]
and either netcat the output with ``nc -l -u -p 2442`` or using
```python
from socket import *
from matplotlib import pyplot as plt
import array
import numpy as np

plt.ion()
plt.show(block = False)
m=0
while True:
    try:
        s = socket(AF_INET, SOCK_DGRAM)
        s.bind(('', 2442))
        s.settimeout(1)
        data, server = s.recvfrom(400)
        s.close()
        m=m+1
        print(len(data))
    except timeout:
        print("timeout")
```
and even as small as N=40 (i.e. 320 bytes, much less than the 1492 maximum) stops after a couple of transferred packets.