gnuradio/gnuradio #7008: gr_modtool: rename only works on C++ blocks

AsriFox, 2023-12-14:
### What happened?

`gr_modtool rename` can't rename blocks, if they don't have a C++ implementation. 

I suspect that [this line](https://github.com/gnuradio/gnuradio/blob/6404f371abb97f282aa1aa270837ffa9f0ddce84/gr-utils/modtool/core/base.py#L37) in `get_block_candidates` should only be applied to `cpp_blocks`.

Other commands are not affected as much, because they can use regular expressions, so they work with Python blocks (e.g. [tx_hier](https://github.com/AsriFox/gr-dvbs2acm/blob/main/python/dvbs2acm/tx_hier.py)), GRC-only blocks (e.g. [modcod_probe](https://github.com/AsriFox/gr-dvbs2acm/blob/main/grc/dvbs2acm_modcod_probe.block.yml)) and header-only "blocks" (e.g. [dvbs2_config](https://github.com/AsriFox/gr-dvbs2acm/blob/main/include/gnuradio/dvbs2acm/dvbs2_config.h)). 

### System Information

OS: Ubuntu 23.10
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

gfac5e1a33

### Steps to Reproduce the Problem

1. Create a Python block;
2. Run `gr_modtool rename` to rename the new block.

### Relevant log output

```shell
~/Documents/gr-dvbs2acm$ gr_modtool rename tx_hier
GNU Radio module name identified: dvbs2acm
ModToolException: Blockname for renaming does not exists!

~/Documents/gr-dvbs2acm$ gr_modtool rename
GNU Radio module name identified: dvbs2acm
Enter name of block/code to rename (without module name prefix): tx_hier
ModToolException: Blockname for renaming does not exists!
```

willcode, 2023-12-14:
Adding something like
`.split('.')[0].split('_impl')[0]`
but preferably more elegant, would work.

AsriFox, 2023-12-14:
Well, seems like the issue is not in that line, but in [another](https://github.com/gnuradio/gnuradio/blob/6404f371abb97f282aa1aa270837ffa9f0ddce84/gr-utils/modtool/core/base.py#L33):
```python
python_blocks = [x for x in glob.glob1("python", "*.py") if not (x.startswith('qa_') or
                 x.startswith('build') or x.startswith('__init__'))]
```
Here the glob is pointing to `python` directory, but all Python-related files are located in `python/modname`, and `glob1` is specifically a *non-recursive* function.

PR #7009 (2024-01-30): gr_modtool: fix non-C++ block detection and manipulation
## Description
[`get_block_candidates`](https://github.com/gnuradio/gnuradio/blob/6404f371abb97f282aa1aa270837ffa9f0ddce84/gr-utils/modtool/core/base.py#L25-L39) function is used in `gr_modtool` to detect blocks in the OOT module. However, the current implementation only detects C++ blocks, because the pattern for Python files does not check the directory recursively.

Also, the `rename` subcommand does not check for non-C++ blocks and throws an exception if there are no C++ implementation files for the selected block.

This PR presents  [... cut]
commit a36fa78a (2024-01-30): gr_modtool: detect and rename Python blocks

`get_block_candidates`: glob for Python files now scans subdirectories
of `python`.
When missing a header, renaming of Python bindings is skipped.

Fixes #7008
commit cca111fa (2024-03-08): gr_modtool: detect and rename Python blocks

`get_block_candidates`: glob for Python files now scans subdirectories
of `python`.
When missing a header, renaming of Python bindings is skipped.

Fixes #7008

(cherry picked from commit a36fa78ad69d018c2dcb9a005eb43900bc788316)
closed 2024-01-30 by willcode