gnuradio/gnuradio #7014: QT GUI RFNoC Fosphor Display block fails to build on 32-bit ARM

argilo, 2023-12-19:
Debian build logs show that 3.10.9.0-rc1 fails to build on the armel & armhf architectures:

https://buildd.debian.org/status/fetch.php?pkg=gnuradio&arch=armel&ver=3.10.9.0%7Erc1-1&stamp=1702974896&raw=0
https://buildd.debian.org/status/fetch.php?pkg=gnuradio&arch=armhf&ver=3.10.9.0%7Erc1-1&stamp=1702973328&raw=0

I confirmed that maint-3.10 also fails to build on 32-bit Raspberry Pi OS.

It appears the problem is that `QRfnocF15ColorMapper` and `QRfnocF15Surface` (added in #6515) use OpenGL 2.0 functions which were deprecated in 2008: https://www.khronos.org/opengl/wiki/History_of_OpenGL#Deprecation_Model

I expect platforms are beginning to drop support for OpenGL 2.0.

Build logs:
```
[1430/1794] /usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQWT_DLL -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -Dgnuradio_qtgui_EXPORTS -I"/<<PKGBUILDDIR>>/gr-qtgui/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/controlport" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/pmt/../../include" -I"/<<PKGBUILDDIR>>/gr-fft/lib/../include" -I"/<<PKGBUILDDIR>>/gr-filter/lib/../include" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/include" -isystem /usr/include/qwt -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem /usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -isystem /usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -O3 -DNDEBUG -std=c++17 -fPIC -fPIC -Winvalid-pch -include "/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/cmake_pch.hxx" -MD -MT gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o -MF gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o.d -o gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o -c '/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc'
FAILED: gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o 
/usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQWT_DLL -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -Dgnuradio_qtgui_EXPORTS -I"/<<PKGBUILDDIR>>/gr-qtgui/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/controlport" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/pmt/../../include" -I"/<<PKGBUILDDIR>>/gr-fft/lib/../include" -I"/<<PKGBUILDDIR>>/gr-filter/lib/../include" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/include" -isystem /usr/include/qwt -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem /usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -isystem /usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -O3 -DNDEBUG -std=c++17 -fPIC -fPIC -Winvalid-pch -include "/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/cmake_pch.hxx" -MD -MT gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o -MF gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o.d -o gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/QRfnocF15ColorMapper.cc.o -c '/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc'
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc: In member function ‘void gr::qtgui::QRfnocF15ColorMapper::drawScale(std::string, float, float, float, float)’:
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:164:15: error: ‘GL_TEXTURE_ENV’ was not declared in this scope; did you mean ‘GL_TEXTURE_2D’?
  164 |     glTexEnvf(GL_TEXTURE_ENV, GL_TEXTURE_ENV_MODE, GL_REPLACE);
      |               ^~~~~~~~~~~~~~
      |               GL_TEXTURE_2D
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:164:31: error: ‘GL_TEXTURE_ENV_MODE’ was not declared in this scope; did you mean ‘GL_TEXTURE_MAX_LOD’?
  164 |     glTexEnvf(GL_TEXTURE_ENV, GL_TEXTURE_ENV_MODE, GL_REPLACE);
      |                               ^~~~~~~~~~~~~~~~~~~
      |                               GL_TEXTURE_MAX_LOD
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:164:5: error: ‘glTexEnvf’ was not declared in this scope
  164 |     glTexEnvf(GL_TEXTURE_ENV, GL_TEXTURE_ENV_MODE, GL_REPLACE);
      |     ^~~~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:167:5: error: ‘glBegin’ was not declared in this scope
  167 |     glBegin(GL_QUADS);
      |     ^~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:168:5: error: ‘glTexCoord2f’ was not declared in this scope
  168 |     glTexCoord2f(0.0f, 0.0f);
      |     ^~~~~~~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:169:5: error: ‘glVertex2f’ was not declared in this scope
  169 |     glVertex2f(x0, y0);
      |     ^~~~~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15ColorMapper.cc:176:5: error: ‘glEnd’ was not declared in this scope
  176 |     glEnd();
      |     ^~~~~
[1431/1794] /usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGR_CTRLPORT -DGR_MPLIB_GMP -DGR_PERFORMANCE_COUNTERS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQWT_DLL -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSPDLOG_SHARED_LIB -Dgnuradio_qtgui_EXPORTS -I"/<<PKGBUILDDIR>>/gr-qtgui/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/../include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/lib/controlport" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/lib/pmt/../../include" -I"/<<PKGBUILDDIR>>/gr-fft/lib/../include" -I"/<<PKGBUILDDIR>>/gr-filter/lib/../include" -I"/<<PKGBUILDDIR>>/gnuradio-runtime/include" -I"/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gnuradio-runtime/include" -isystem /usr/include/qwt -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -isystem /usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -isystem /usr/include/arm-linux-gnueabihf/qt5/QtOpenGL -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -fstack-clash-protection -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -O3 -DNDEBUG -std=c++17 -fPIC -fPIC -Winvalid-pch -include "/<<PKGBUILDDIR>>/obj-arm-linux-gnueabihf/gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/cmake_pch.hxx" -MD -MT gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/rfnoc_f15_display_impl.cc.o -MF gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/rfnoc_f15_display_impl.cc.o.d -o gr-qtgui/lib/CMakeFiles/gnuradio-qtgui.dir/rfnoc_f15_display_impl.cc.o -c '/<<PKGBUILDDIR>>/gr-qtgui/lib/rfnoc_f15_display_impl.cc'
[... cut ...]
      |     ^~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15Surface.cc:406:5: error: ‘glTexCoord2d’ was not declared in this scope
  406 |     glTexCoord2d(0.0f, 0.0f);
      |     ^~~~~~~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15Surface.cc:407:5: error: ‘glVertex2d’ was not declared in this scope
  407 |     glVertex2d(x[0], y[0]);
      |     ^~~~~~~~~~
/<<PKGBUILDDIR>>/gr-qtgui/lib/QRfnocF15Surface.cc:414:5: error: ‘glEnd’ was not declared in this scope
  414 |     glEnd();
      |     ^~~~~
```

argilo, 2023-12-19:
The problematic code was removed from maint-3.10 in #7017, but it remains on the main branch. @mbr0wn is planning to look at it soon.

mbr0wn, 2024-01-22:
Note: We aren't using that many GL calls directly, mostly it's going through Qt. However, we are using classes that are officially deprecated/obsolete! Classes like `QGLWidget` are replaced by `QOpenGLWidget` (all `QGL*` are replaced by `QOpenGL*` classes). However, they're not 1:1 replacements. I still think updating the Qt widgets are the right place to start.