gnuradio/gnuradio #7021: Circular buffers are broken on sparc64

argilo, 2023-12-20:
Debian build logs for the sparc64 architecture show a large number of test failures, because GNU Radio's circular buffers don't work:

https://buildd.debian.org/status/logs.php?pkg=gnuradio&arch=sparc64

For 3.10.9.0~rc1-1 the failure was:
```
RuntimeError: gr::vmcircbuf_sysconfig
```

For 3.7.10.1-2 the failure was:
```
vmcircbuf_createfilemapping: createfilemapping is not available
gr::vmcircbuf_sysv_shm: shmat (3): Invalid argument
gr::vmcircbuf_sysv_shm: shmat (3): Invalid argument
gr::vmcircbuf_sysv_shm: shmat (3): Invalid argument
terminate called after throwing an instance of 'gr::signal'
Aborted
```

argilo, 2023-12-20:
From the logs on GNU Radio 3.7, it would appear that the third `shmat` call in the `vmcircbuf_sysv_shm` constructor returns `EINVAL`:

https://github.com/gnuradio/gnuradio/blob/cce52e9dfd1ea6e0a081e824754422852e0872b1/gnuradio-runtime/lib/vmcircbuf_sysv_shm.cc#L109

argilo, 2023-12-20:
From the linux man page, `EINVAL` means:

> Invalid `shmid` value, unaligned (i.e., not page-aligned and `SHM_RND` was not specified) or invalid `shmaddr` value, or can't attach segment at `shmaddr`, or `SHM_REMAP` was specified and `shmaddr` was `NULL`.

argilo, 2023-12-20:
Tests failed the same way on hppa, prior to Debian bypassing them:

https://buildd.debian.org/status/logs.php?pkg=gnuradio&arch=hppa

argilo, 2023-12-26:
I installed Debian in qemu-system-sparc64 and built gnuradio-runtime. With debug logging enabled, `runtime_qa_vmcircbuf.cc` produces the following output:
```
$ gnuradio-runtime/lib/runtime_qa_vmcircbuf.cc
Running 1 test case...
standard_tests :info: Testing gr::vmcircbuf_createfilemapping_factory...
pagesize :debug: Setting pagesize to 8192 B
gr::vmcircbuf :info: vmcircbuf_createfilemapping: CreateFileMapping is not available
gr::test_a_bunch :info: Failed to allocate gr::vmcircbuf number 1 of size 8192 (cum = 0)
standard_tests :info: gr::vmcircbuf_createfilemapping_factory: Doesn't work
test all factories :debug: testing gr::vmcircbuf_createfilemapping_factory: false
standard_tests :info: Testing gr::vmcircbuf_sysv_shm_factory...
gr::vmcircbuf :error: shmat (3): Invalid argument
gr::vmcircbuf :error: shmat (3): Invalid argument
gr::vmcircbuf :error: shmat (3): Invalid argument
gr::test_a_bunch :info: Failed to allocate gr::vmcircbuf number 1 of size 8192 (cum = 0)
standard_tests :info: gr::vmcircbuf_sysv_shm_factory: Doesn't work
test all factories :debug: testing gr::vmcircbuf_sysv_shm_factory: false
standard_tests :info: Testing gr::vmcircbuf_mmap_shm_open_factory...
gr::vmcircbuf :error: mmap (2) failed
gr::test_a_bunch :info: Failed to allocate gr::vmcircbuf number 1 of size 8192 (cum = 0)
standard_tests :info: gr::vmcircbuf_mmap_shm_open_factory: Doesn't work
test all factories :debug: testing gr::vmcircbuf_mmap_shm_open_factory: false
standard_tests :info: Testing gr::vmcircbuf_mmap_tmpfile_factory...
gr::vmcircbuf :error: mmap (2) failed
gr::test_a_bunch :info: Failed to allocate gr::vmcircbuf number 1 of size 8192 (cum = 0)
standard_tests :info: gr::vmcircbuf_mmap_tmpfile_factory: Doesn't work
test all factories :debug: testing gr::vmcircbuf_mmap_tmpfile_factory: false
/home/argilo/git/gnuradio/gnuradio-runtime/lib/qa_vmcircbuf.cc(22): fatal error: in "test_all": critical check gr::vmcircbuf_sysconfig::test_all_factories(verbose) has failed
```

argilo, 2023-12-27:
I tracked down the bug. GNU Radio aligns shared memory to page boundaries, but some architectures have stricter alignment requirements.

The documentation for `shmat` (used by `gr::vmcircbuf_sysv_shm_factory`) states:

> `SHMLBA`
> 
> Segment low boundary address multiple. When explicitly specifying an attach address in a call to `shmat()`, **the caller should ensure that the address is a multiple of this value. This is necessary on some architectures**, in order either to ensure good CPU cache performance or to ensure that different attaches of the same segment have consistent views within the CPU cache. `SHMLBA` is normally some multiple of the system page size. (On many Linux architectures, SHMLBA is the same as the system page size.)

And the documentation from `mmap` (used by `gr::vmcircbuf_mmap_shm_open_factory` and `gr::vmcircbuf_mmap_tmpfile_factory`) states:

> `MAP_FIXED`
>
> Don't interpret `addr` as a hint: place the mapping at exactly that address. **`addr` must be suitably aligned: for most architectures a multiple of the page size is sufficient; however, some architectures may impose additional restrictions.** If the memory region specified by `addr` and `len` overlaps pages of any existing mapping(s), then the overlapped part of the existing mapping(s) will be discarded. If the specified address cannot be used, `mmap()` will fail.
>
> Software that aspires to be portable should use the `MAP_FIXED` flag with care, keeping in mind that the exact layout of a process's memory mappings is allowed to change significantly between kernel versions, C library versions, and operating system releases. Carefully read the discussion of this flag in NOTES!

argilo, 2023-12-27:
On sparc64 specifically, the page size is 8192 while `SHMLBA` is 16384. If I change `gr::pagesize` to return `SHMLBA` then `qa_vmcircbuf` passes, and the logs show that all three `vmcircbuf` implementations work.

willcode, 2023-12-27:
Changing to SHMLBA sounds painless. Thanks!

argilo, 2023-12-27:
> Changing to `SHMLBA` sounds painless.

There's a bit of nuance because the `pagesize` and `vmcircbuf` code is platform-specific, but I don't expect it will be hard to increase the alignment to `SHMLBA` on platforms that have it.

argilo, 2023-12-27:
Another thing to watch out for is that `SHMLBA` is large on some architectures:

```
glibc-2.38/sysdeps/unix/sysv/linux/bits/shmlba.h
26:#define SHMLBA		(__getpagesize ())

glibc-2.38/sysdeps/unix/sysv/linux/sparc/bits/shmlba.h
26:#define SHMLBA		(__getshmlba ())

glibc-2.38/sysdeps/unix/sysv/linux/loongarch/bits/shmlba.h
24:#define SHMLBA		0x10000

glibc-2.38/sysdeps/unix/sysv/linux/mips/bits/shmlba.h
24:#define SHMLBA		0x40000

glibc-2.38/sysdeps/unix/sysv/linux/arm/bits/shmlba.h
26:#define SHMLBA		(__getpagesize () << 2)

glibc-2.38/sysdeps/unix/sysv/linux/ia64/bits/shmlba.h
24:#define SHMLBA		(1024 * 1024)

glibc-2.38/sysdeps/unix/sysv/linux/csky/bits/shmlba.h
26:#define SHMLBA		(__getpagesize () << 2)

glibc-2.38/sysdeps/unix/sysv/linux/sh/bits/shmlba.h
24:#define SHMLBA		0x4000

glibc-2.38/sysdeps/unix/sysv/linux/hppa/bits/shmlba.h
24:#define SHMLBA		0x1000
```

On IA64 it's 1048576, and on MIPS it's 262144. I would guess that on those architectures it's not a hard requirement, and that using `SHMLBA` as alignment would bloat the circular buffers.

willcode, 2023-12-27:
Also, if nobody is using a platform and we're not going to test on it, then maybe we explicity unsupport the platform.

argilo, 2023-12-27:
Debian runs our test suite on a ton of architectures, so I figured we may as well use the opportunity to rid GNU Radio of architecture-specific bugs. There are very few things in GNU Radio that are architecture-specific, and this bug appears to be the last one causing test failures.

argilo, 2023-12-28:
I had a peek behind the curtain (in the Linux kernel) to see what the situation is for various architectures.

As I suspected, a several of architectures have `SHMLBA` set higher than `PAGE_SIZE` for efficiency reasons, but still allow mappings with page alignment. Other architectures enforce `SHMLBA` alignment in `shmat` (typically by defining the `__ARCH_FORCE_SHMLBA` macro), but not all of these architectures enforce the same alignment in `mmap`.

PA-RISC / HPPA is a special case. It only requires page alignment, but for pages that are mapped to the same physical address (which GNU Radio needs for its circular buffers) the two addresses must be equal modulo 0x400000.

The table below summarizes the situation, with ✅ indicating situations where GNU Radio's current code will work, and ❌ indicating where it will not.

| Architecture | shmat granularity | mmap granularity |
| ----- | ----- | ----- |
| alpha | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| arc | ❌ SHMLBA | ❌ SHMLBA |
| arm | ❌ SHMLBA | ✅ PAGE_SIZE, if cache is VIVT<br>❌ SHMLBA, if cache is VIPT |
| arm64 | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| csky | ❌ SHMLBA | ❌ SHMLBA |
| hexagon | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| ia64 | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| loongarch | ✅ PAGE_SIZE, if page size is 64 kB<br>❌ SHMLBA, if page size is 4 kB or 16 kB | ✅ PAGE_SIZE, if page size is 64 kB<br>❌ SHMLBA, if page size is 4 kB or 16 kB |
| m68k | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| microblaze | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| mips | ❌ SHMLBA | ✅ PAGE_SIZE |
| nios2 | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| openrisc | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| parisc (hppa) | ❌ PAGE_SIZE, but pages mapped to<br>same physical address must have equal<br>addresses modulo 0x400000 | ❌ PAGE_SIZE, but pages mapped to<br>same physical address must have equal<br>addresses modulo 0x400000 |
| powerpc | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| riscv | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| s390 | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| sh | ❌ SHMLBA | ✅ PAGE_SIZE |
| sparc | ❌ SHMLBA | ❌ SHMLBA |
| um | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| x86 | ✅ PAGE_SIZE | ✅ PAGE_SIZE |
| xtensa | ❌ SHMLBA | ❌ SHMLBA |

argilo, 2023-12-28:
I think the following initialization strategy would allow GNU Radio to function on many more architectures (all except PA-RISC) while avoiding buffer bloat:

1. Test all available circular buffer implementations with `PAGE_SIZE` granularity.
2. If none are working, and `SHMLBA` differs from `PAGE_SIZE`, then test all available circular buffer implementations with `SHMLBA` granularity.

The current initialization strategy only executes step 1.

argilo, 2023-12-28:
Also, I wonder if GNU Radio could fall back to the single-mapped `host_buffer` if none of the double-mapped `vmcircbuf` implementations work.

marcusmueller, 2023-12-28:
Wow, that is some solid investigations! I wonder whether on these architectures where we take a performance hit only if we use pagesize instead of shmlba it's actually *worth it*, seeing that instead of working on something that might reasonably fit L2 caches, we're making our buffer larger. Then again, we won't be able to benchmark that on any machine that we have sitting under our desks…

so yeah, assuming our target audience is mainly

1. scientists  sitting in front of HPC clusters
2. embedded folks porting GNU Radio to deeply embedded devices of funky low-cost architectures

I think a warning that says `SHMLBA != getpagesize: there might be performance degradations` would be optimal for group 1; they are running parallelized software on NUMA systems, they hopefully have more in-house experts on these machines than we do. And: If someone needs their sunway cluster to run GNU Radio at benchmarked max performance, patches are welcome… 

For group 2: taking elaborate guesses what makes these chips happy, and how deep our support needs to run at all (xtensa, mips, csky, nios2, microblaze, m68k, even arm, are all 32 bit, not typically DSP-heavy, though the softcores might be interesting as accelerator orchestrators) is probably a bit in vain; the configurations that we might encounter in the wild will have MMU features enabled or disabled, different memory bandwidths… 

So I'd say that your simple "try pagesize, then shmlba" seems reasonable.

argilo, 2023-12-28:
> Also, I wonder if GNU Radio could fall back to the single-mapped `host_buffer` if none of the double-mapped `vmcircbuf` implementations work.

I did a quick test by uncommenting `#define FORCE_SINGLE_MAPPED` in io_signature.h, and this appears to be feasible. However, when running the test suite, I quickly ran into bugs:

1. `agc3_cc` overruns its input and output buffers, because it calls `set_output_multiple` and expects `noutput_items` to be a multiple of that value, but then calls `set_alignment`, which blows away the output multiple.
2. Tagged Stream blocks do not appear to be compatible with custom buffers, because they refuse to produce output until there's enough room for a PDU, while custom buffers don't provide additional room until the block has used up what it already has. This bad behaviour of Tagged Stream blocks already causes other problems for the scheduler, as noted in #7028.

argilo, 2023-12-28:
Additionally, dynamic flowgraph reconfiguration results in `std::runtime_error: Unexpected transfer type for host_buffer: DEFAULT_INVALID`, and shared pointer reference count checks fail in the `t12_release_shared_pointers` test case:

```
$ gr-blocks/lib/blocks_qa_gr_top_block.cc
Running 13 test cases...
qa_top_block::t0()
qa_top_block::t1()
qa_top_block::t2()
qa_top_block::t3()
qa_top_block::t4()
host_buffer :error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
unknown location(0): fatal error: in "t4_reconfigure": std::runtime_error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(91): last checkpoint: "t4_reconfigure" test entry
qa_top_block::t5()
qa_top_block::t6()
host_buffer :error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
unknown location(0): fatal error: in "t6_reconfig_max_noutputs": std::runtime_error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(136): last checkpoint: "t6_reconfig_max_noutputs" test entry
qa_top_block::t7()
qa_top_block::t8()
host_buffer :error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
unknown location(0): fatal error: in "t8_reconfig_max_noutputs_per_block": std::runtime_error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(184): last checkpoint: "t8_reconfig_max_noutputs_per_block" test entry
qa_top_block::t9()
qa_top_block::t10()
host_buffer :error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
unknown location(0): fatal error: in "t10_reconfig_max_output_buffer": std::runtime_error: Unexpected transfer type for host_buffer: DEFAULT_INVALID
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(234): last checkpoint: "t10_reconfig_max_output_buffer" test entry
qa_top_block::t12()
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(312): error: in "t12_release_shared_pointers": check 1 == src.use_count() has failed [1 != 2]
/home/argilo/prefix_311/src/gnuradio/gr-blocks/lib/qa_gr_top_block.cc(313): error: in "t12_release_shared_pointers": check 1 == head.use_count() has failed [1 != 2]

*** 6 failures are detected in the test module "Master Test Suite"
```