gnuradio/gnuradio #7026: GRC: Blocks overlap on HiDPI screens

haakov, 2023-12-21:
### What happened?

I am running GRC on a 2880x1800 14-inch laptop. I have set `Xft.dpi = 144` in my .Xresources, which I believe leads to a scaling factor of 1.5. The default flowgraph looks like this:
[image]

If I remove `Xft.dpi = 144` from .Xresources, the blocks look better:
[image]

This issue is present both in GRC Gtk and GRC Qt.

### System Information

OS: Arch Linux
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Open the same flowgraph with different DPI settings.

willcode, 2023-12-21:
In `grc/gui/Constants.py` there is `DPI_SCALING`. This is fixed at `1.0` at the moment. For a source install, you can edit this value. For a package-based install, if you really want to, you can change the value at the installed location. For example, on my source-based install, this file ends up at

`<prefix>/lib64/python3.11/site-packages/gnuradio/grc/gui/Constants.py`

It would be a relatively simple matter to have GRC read this value from the config file.

HACKER097, 2024-01-13:
Hello, I would like to be assigned this issue. I think it can be solved by added these lines to `grc/gui/Constants.py`

```python
XRESOURCES_CONFIG_PATH = "~/.Xresources"

def update_dpi_scaling():
    global DPI_SCALING
    
    file_path = os.path.expanduser(XRESOURCES_CONFIG_PATH)

    key = "Xft.dpi"

    with open(file_path, 'r') as file:
        for line in file:
            line = line.strip()

            if key in line:
                parts = line.split(':')
                dpi = int(parts[1].strip())
                
                # SCALING_FACTOR = GIVEN_DPI/STANDARD_DPI
                DPI_SCALING = dpi/96

update_dpi_scaling()
```

I do feel like `Constants.py` is not the best place to be reading files, and that I should be using a parser (configparser does not work with `.Xresources`) instead of doing a manually.  But I couldn't find a more apt location to add this, or a parser that works with `.Xresources`.

Please let me know if this is an acceptable solution.

willcode, 2024-01-13:
I don't think we want to look at `~/.Xresources`. There are ways to get DPI via Gtk/Qt, and then any additional scaling factor (sometimes an app still want its own scaling) would go in `~/.gnuradio/grc.conf`.

HACKER097, 2024-01-13:
> There are ways to get DPI via Gtk/Qt

Wouldn't that just be a matter of uncommenting these lines?

```python
# _SCREEN = Gdk.Screen.get_default()
# _SCREEN_RESOLUTION = _SCREEN.get_resolution() if _SCREEN else -1
# DPI_SCALING = _SCREEN_RESOLUTION / 96.0 if _SCREEN_RESOLUTION > 0 else 1.0
```

willcode, 2024-01-13:
Likely that didn't work in all cases, so it got commented out. I don't know if GTK recognizes fractional scaling.

jason-merlo, 2024-04-06:
It's unclear if this is related or a separate issue, but on macOS I'm having an issue with _different_ scaling between GRC GTK and QT, which prevents me from being able to test GRC QT much since I collaborate with others using GRC GTK and have GTK on some of my machines. Since migration to GRC QT as the default is now on the horizon I wanted to make sure this was being tracked.

If this is a separate issue, I can open a new issue.  I've attached some screenshots of a simple flow graph here running on a 163dpi monitor. A similar issue occurs on a 109dpi monitor.

[image]
[image]

willcode, 2024-04-06:
Under File/Preferences/Appearance, scaling can be set (requires a restart). There may also be system scaling happening. For me, overlapping often happens when opening someone else's flowgraph under Gtk. Currently, using Gtk and Qt on my Linux HiDPI machine results in the same layout. So, this is probably related to Qt only in that scaling is set up differently on your machine for Gtk and Qt.

jason-merlo, 2024-04-06:
It seems the "QT scale factor" scales the _entire_ UI for me, not just for canvas item placement.  The UI scaling seems correct to me at the default of 1.0.  I think the only issue I'm observing is item placement on the canvas between QT and GTK is not consistent.  Is there a different scaling factor used for canvas placement?

Attached is a "QT Scale factor = 2.0" for the same flow graph above for reference.  Note that it is identical to the one above, but the native window title bar is smaller (for scale reference).

[image]

marcusmueller, 2024-12-26:
@jason-merlo are you still working on this?

Gowri213, 2025-12-28:
@marcusmueller I’ve been reading through this issue and the related PRs (#7545 and #7546), which looked close but were closed due to branch deletion.

I’d like to pick this up if this is open

Pranav-0440, 2026-10-19:
@haakov /attempt

harshsrah18, 2026-10-21:
Hello, I’d like to work on this issue. I think it may be possible to address the HiDPI scaling problem by adding the necessary DPI scaling logic to `grc/gui/Constants.py`.

I’d be happy to investigate the issue, implement the fix, and test it. Could you please assign this issue to me?  

import os

XRESOURCES_CONFIG_PATH = "~/.Xresources"

def update_dpi_scaling():
    global DPI_SCALING

    file_path = os.path.expanduser(XRESOURCES_CONFIG_PATH)
    DPI_SCALING = 1.0

    try:
        with open(file_path, "r", encoding="utf-8") as file:
            for line in file:
                line = line.split("!", 1)[0].strip()

                if not line.startswith("Xft.dpi"):
                    continue

                parts = line.replace("=", ":", 1).split(":", 1)
                if len(parts) != 2 or parts[0].strip() != "Xft.dpi":
                    continue

                try:
                    dpi = float(parts[1].strip())
                    if dpi > 0:
                        DPI_SCALING = dpi / 96.0
                        return
                except ValueError:
                    continue
    except OSError:
        pass

DPI_SCALING = 1.0
update_dpi_scaling()