gnuradio/gnuradio #7029: Textboxes close to unreadable on dark mode

electron271, 2023-12-27:
### What happened?

Using Breeze Dark textboxes are close to unreadable (white on white)
[image]

### System Information

OS: Arch Linux
GR Installation Method: pacman

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0

### Steps to Reproduce the Problem

- Launch GNU Radio Companion with dark mode on
- Right click on a node
- Look at the unreadable text

electron271, 2023-12-27:
Also should note my settings.ini for gtk-3.0

```ini
[Settings]
gtk-theme-name=Breeze-Dark
gtk-icon-theme-name=breeze-dark
gtk-font-name=FiraCode Nerd Font Light 11
gtk-cursor-theme-name=Adwaita
gtk-cursor-theme-size=0
gtk-toolbar-style=GTK_TOOLBAR_BOTH
gtk-toolbar-icon-size=GTK_ICON_SIZE_LARGE_TOOLBAR
gtk-button-images=1
gtk-menu-images=1
gtk-enable-event-sounds=1
gtk-enable-input-feedback-sounds=1
gtk-xft-antialias=1
gtk-xft-hinting=1
gtk-xft-hintstyle=hintfull
gtk-application-prefer-dark-theme = true
```

willcode, 2023-12-27:
See #6888, #6942.

electron271, 2023-12-27:
It seems possible that for whatever reason the code is not detecting dark theme.

I have also had the issue of foliate not using dark theme, so that might be related.

marcusmueller, 2023-12-27:
As @willcode  said, this is already fixed in the next release

closed 2023-12-27 by marcusmueller