gnuradio/gnuradio #7033: set_output_multiple and set_alignment cannot be used together

argilo, 2023-12-28:
`set_output_multiple` is documented as:

> Constrain the `noutput_items` argument passed to `forecast` and `general_work`
>
> `set_output_multiple` causes the scheduler to ensure that the `noutput_items` argument passed to `forecast` and `general_work` will be an integer multiple of `multiple`. The default value of output multiple is 1.

`set_alignment` is documented as:

> Constrains buffers to work on a set item alignment (for SIMD)
> 
> `set_alignment_multiple` **[sic]** causes the scheduler to ensure that the `noutput_items` argument passed to `forecast` and `general_work` will be an integer multiple of `multiple`. The default value is 1.
>
> This control is similar to the `output_multiple` setting, except that if the number of items passed to the block is less than the `output_multiple`, this value is ignored and the block can produce like normal. The `d_unaligned` value is set to the number of items the block is off by. In the next call to `general_work`, the `noutput_items` is set to `d_unaligned` or less until `d_unaligned==0`. The buffers are now aligned again and the aligned calls can be performed again.

Under the covers, both of these methods set the `d_output_multiple` instance variable:

https://github.com/gnuradio/gnuradio/blob/baf51f5e381ee3b4a7bd8a4ea4969135a75c2ac6/gnuradio-runtime/lib/block.cc#L106-L113

https://github.com/gnuradio/gnuradio/blob/baf51f5e381ee3b4a7bd8a4ea4969135a75c2ac6/gnuradio-runtime/lib/block.cc#L115-L121

The only difference is that `set_output_multiple` also sets `d_output_multiple_set` to `true`, which tells the block executor that it not allowed to provide fewer than `output_multiple` items when the number of available items is too small.

Because these methods both set `d_output_multiple`, they cannot be used together. Unfortunately, this has led to problems on at least two occasions:

1. https://github.com/gnuradio/gnuradio/blob/4d3eb9f321bf28d7d6310e304a6a0323cdfa66aa/gr-digital/lib/corr_est_cc_impl.cc#L86-L91
2. #7032

To prevent this situation from causing bugs, I can think of a few possible solutions:

1. Refactor `set_output_multiple` and `set_alignment` so they can be used together.
2. Have `set_output_multiple` and `set_alignment` raise an exception if they are used together.
3. Document that `set_output_multiple` and `set_alignment` cannot be used together.

argilo, 2023-12-28:
I think my preference would be to implement solutions 2 & 3. I'm not sure there's much sense in allowing both an output multiple and an alignment to be set at the same time.

argilo, 2023-12-28:
Another for the list:

4. Log a warning if `set_output_multiple` and `set_alignment` are used together.

Compared to solution 2, this wouldn't be as easy to notice, but it also wouldn't carry the risk of breaking existing code.

willcode, 2023-12-28:
Is the only different between the two that `alignment` is a "please" and `multiple` is a "have to"? It's pretty confusing in the executor what would even be desired if they were both set.

How about unsetting `d_output_multiple_set` in `set_alignment()` and documenting in the API that only the last call has an effect? It's not great, but it's also not much work, and is already what happens if `set_output_multiple()` is called after `set_output_alignment()`.

argilo, 2023-12-28:
> Is the only different between the two that `alignment` is a "please" and `multiple` is a "have to"?

Pretty much. In the `alignment` case, you're giving the scheduler permission to temporarily deviate from providing `n * multiple` samples, which allows the block to get a SIMD performance boost most of the time, while still allowing it to process small batches of samples if the need arises.

argilo, 2023-12-28:
With `output_multiple` the block might sometimes be forced to wait for more samples to arrive from upstream, and will be unable to process tail-end samples if the number isn't an even multiple.

argilo, 2023-12-28:
Given those limitations, it's probably not a great idea for AGC3 to be using `set_output_multiple` for its own convenience, since it imposes upon the scheduler and the rest of the flow graph.

argilo, 2023-12-28:
> How about unsetting `d_output_multiple_set` in `set_alignment()` and documenting in the API that only the last call has an effect? It's not great, but it's also not much work, and is already what happens if `set_output_multiple()` is called after `set_output_alignment()`.

Sure, we could consider that as potential solution # 5.

Like solution 1, it could change the behaviour of existing blocks. That may be warranted, since blocks that call both methods are likely to be buggy.

But one downside of solution 5 is that it could cause a delayed & dangerous failure (e.g. buffer overflow) for a block whose `work` function expects to always receive `n * multiple` samples (as AGC3's did) versus an immediate and safe failure (exception) with solution 1.