gnuradio/gnuradio #7041: Null Source doesn't produce output

argilo, 2024-01-03:
While experimenting with changes to the block executor to detect misbehaving blocks, I noticed that the Null Source doesn't actually produce any output:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-blocks/lib/null_source_impl.cc#L36-L46

Its work loop iterates over `input_items`, which is empty, so the `memset` never executes.

It appears this bug was introduced when Null Source was extended to support multiple output streams: https://github.com/gnuradio/gnuradio/commit/92b01e30373e6ed36d1067e2bcba661d2253f34a#diff-b837595e779a0fce1911a02fd8039a2ad1a82612f30efc6fbb0f11c31a3c2e40

The only reason the Null Source can get away with this is that the scheduler initializes all buffers to zero. (I suspect this happens implicitly when shared memory is allocated.)

I can see two paths forward:

1. Change the Null Source's loop to iterate over `output_items`. This would guarantee that the block works regardless of the scheduler's buffer initialization behaviour, but would increase the CPU utilization of the Null Source block.
2. Declare that the scheduler is responsible for initializing all buffers to zero, and change `null_source_impl::work` to a no-op. This would keep the Null Source's CPU utilization low, but would place an additional requirement on all scheduler implementations. (It's probably a reasonable requirement, though.)

Any opinions on which of these options is better?

daniestevez, 2024-01-03:
I prefer option 2, because of the performance benefit. I don't see a reason for not initializing all the buffers to zero at the start of the flowgraph, and given that the producer is the only object that should write to a buffer, the null source could assume that its output buffer will always be filled with zeros without the need for a `memset()`.

mbr0wn, 2024-01-08:
Second that.

willcode, 2024-01-08:
This will work for typical CPU buffers. @jsallay @mormj will this hold true for custom buffers in general, i.e., is underlying memory always permanently dedicated to a block?

jsallay, 2024-01-08:
The example cuda_buffers: https://github.com/BlackLynx-Inc/gr-cuda_buffer/blob/main/lib/cuda_buffer.cc doesn't initialize the memory to 0, but it could be changed to do so.  If we make it a requirement, then I think we would need to push a change to that repo.  (I don't know how hard that would be).

I think I could go either way for (1) or (2).  I agree with the performance consideration already brought up.  My concern, in particular with custom buffers, is that this is a feature that would probably only be used by a single block.  If I am writing a custom buffer, I need to know to add the behavior in and to test it against a null_source.

willcode, 2024-01-08:
I'm going to vote for (1) since I don't think blocks should assume their output buffers are 0. It makes sense for a null source, but what about for some other block that thinks it can sparsely write to its output?

argilo, 2024-01-08:
OK, now that I've seen an existing buffer implementation that does not initialize the buffer to zero, I'd prefer (1) also.

argilo, 2024-01-08:
Fix: #7071

closed 2024-01-08 by willcode

daniestevez, 2024-01-09:
When I voted for option 2, I forgot to consider custom buffers. Still, having a requirement to zero-initialize buffers sounds somewhat reasonable to me. I'm also coming at this from a Rust mindset, in which uninitialized data is a potential cause of undefined behaviour and thus frowned upon. However I've also been guilty of not initializing buffers in Rust for performance reasons, particularly in the case of H2D and D2H buffers when doing CUDA.

The main reason we had for preferring option 2 was performance. But in the context of custom buffers (for instance for CUDA) this raises more questions. If we connect a null source to a CUDA block, would only the host-side part of the buffer required to be zero-initialized? Or also the device-side? If both sides are zero initialized, then the H2D copy that happens before calling the CUDA block could be omitted, but how would the runtime know about this special case?

In an ideal world, having the input of a block connected to a null source should be a no-op. But this seems a very special case that is tricky to achieve with custom buffers. So with all this in mind I now prefer option 1 too, as it reduces the chances of unexpected breakage because some custom buffer decided not to zero-initialize the buffer or to move it around in memory.

PR #7071 (2024-01-08): blocks: Fix loop bound in Null Source
## Description
In 92b01e30373e6ed36d1067e2bcba661d2253f34a, the Null Source was extended to support multiple outputs. But its `work` function iterates over inputs instead of outputs, so it doesn't write any output. It only gets away with this because double-mapped buffers are initialized to zero. But as noted in https://github.com/gnuradio/gnuradio/issues/7041#issuecomment-1881100761, this is not true of all buffer implementations, so Null Sink should write zeroes to its outputs.

## Related Issue
Fixes #7041.

## Which blocks/areas does this affect?
* Nul [... cut]