gnuradio/gnuradio #7044: BER Curve generator block broken

marcusmueller, 2024-01-03:
### What happened?

Can't initialize GRC QT GUI Bercurve Sink block

### System Information

OS: F38
GR Installation Method: source

### GNU Radio Version

3.11-git (main)

### Specific Version

mian as of 2024-01-03

### Steps to Reproduce the Problem

open ber_curve_gen example

### Relevant log output

```shell
Param - Bus_Structure_Sink(bus_structure_sink):
	Value "list(map(lambda b:list(map(lambda a:b * len(esno) * 2 + a, range(len(esno)*2))), range(num_curves)))" cannot be evaluated:
	name 'num_curves' is not defined
```

argilo, 2024-01-03:
This looks to be a duplicate of #5627.

argilo, 2024-01-07:
Closing as a duplicate of #5627.

closed 2024-01-07 by argilo