gnuradio/gnuradio #7048: Polyphase Clock Sync segfaults due to missing bounds checks

argilo, 2024-01-03:
As noted in https://github.com/gnuradio/gnuradio/pull/7040#issuecomment-1874818741, Polyphase Clock Sync readily segfaults if a large input (e.g. Noise Source, Amplitude = 1000) is given.

`d_error` ends up being a very large value here:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L426-L430

This causes `d_k` to be large:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L432-L438

That subsequently causes `count` to be large:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L374-L388

That causes a read outside the input array, generating a segfault:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-digital/lib/pfb_clock_sync_ccf_impl.cc#L390

I did a quick test, and the crash occurred with `count = -67141927`.

willcode, 2024-01-05:
Should this be closed (7040 merged). 7048 and 7049 document the other issues. I don't think anything documents Andy's comment that the block needs more changes but that will make it equivalent to the newer sync block.

argilo, 2024-01-05:
> Should this be closed

No, this is a separate bug that I discovered while working on #7040, and it's still present.

argilo, 2024-01-05:
Or were you just asking whether it's worth tracking this separately, given that Polyphase Clock Sync is superseded by Symbol Sync, and on track to be deprecated?