gnuradio/gnuradio #7049: Deprecate Polyphase Clock Sync

argilo, 2024-01-03:
The Symbol Sync block created in #1294 was intended to replace Clock Recovery MM, MSK Timing Recovery, and Polyphase Clock Sync. The first two are now deprecated, but Polyphase Clock Sync is not.

Polyphase Clock Sync has some serious bugs (e.g. #7040 and #7048). It seems from discussion in #7040 that @daniestevez and @awalls-cx18 would be in favour of deprecating it now.

willcode, 2024-01-05:
Deprecating it now would be fine. It's not like deprecated blocks are actually being removed at any detectable rate these days.

daniestevez, 2024-01-07:
Can you share the settings you use for Polyphase Clock Sync? I can then give you what should be equivalent with Symbol Sync, and we can debug further if this doesn't work. A transmitter + receiver flowgraph where both Polyphase Clock Sync and Symbol Sync can be tested would be ideal.

awalls-cx18, 2024-01-07:
The PFB Clock Sync block uses the y[n]y'[n] ML approximation TED, IIRC.  It also initializes damping factor on the loop to be something unexpected:

```
 // Set the damping factor for a critically damped system
    d_damping = 2 * d_nfilters;
```

duggabe, 2024-01-07:
@daniestevez:  The QPSK tutorial https://wiki.gnuradio.org/index.php?title=QPSK_Mod_and_Demod#Decoding has the complete flowgraph as well as the source .grc file.
Most of my testing was with a direct substitution of the Clock Sync with the Symbol Sync.
FYI, the Symbol Sync worked well in testing for BPSK.

daniestevez, 2024-01-08:
@duggabe Here is the version of that flowgraph using Symbol Sync:

[Qpsk_stage6_symbol_sync.grc.zip](https://github.com/gnuradio/gnuradio/files/13857645/Qpsk_stage6_symbol_sync.grc.zip)

I did pretty much a direct substitution, and I didn't adjust the parameters to try to mimic the quirks of the PFB Clock Sync as accurately as possible (as Andy was mentioning in https://github.com/gnuradio/gnuradio/issues/7049#issuecomment-1880130797). I didn't do thorough testing, but I did a quick comparison of the flowgraph with Symbol Sync and the flowgraph with PFB Clock Sync across a range of SNRs, and qualitatively both seem similar. Interestingly, with zero noise the Symbol Sync gives a cleaner constellation.

duggabe, 2024-01-08:
@daniestevez Thank you for those settings :) With that performance I can support deprecating the PFB Clock Sync.
Just for completeness, would the settings for BPSK be the same? re: https://wiki.gnuradio.org/index.php?title=Simulation_example:_BPSK_Demodulation#Decoding
I will plan on updating the QPSK and BPSK tutorials to use the Symbol Sync.

awalls-cx18, 2024-01-08:
Slide 30 of this brief is the comparison of the performance of the two block with zero noise.  I littery just modified the QPSK tutorial for that screen shot.

[Link to presentation](https://www.gnuradio.org/grcon/grcon17/presentations/symbol_clock_recovery_and_improved_symbol_synchronization_blocks/Andy-Walls-Samples-to-Digital-Symbols.pdf)

daniestevez, 2024-01-09:
> Just for completeness, would the settings for BPSK be the same?

@duggabe Not exactly the same settings, since in the BPSK flowgraph you don't have an equalizer and so have output samples / symbol = 1 instead of 2. But this is the only setting that needs to be changed.

Here is the BPSK flowgraph with PFB Clock Sync replaced by Symbol Sync:

[Bpsk_stage6_symbol_sync.grc.zip](https://github.com/gnuradio/gnuradio/files/13871399/Bpsk_stage6_symbol_sync.grc.zip)

marcusmueller, 2024-03-11:
tagging high, since small step and impact on severity of bugs in PFB clock itself