gnuradio/gnuradio #7050: Use-after-free in most QT GUI sinks

argilo, 2024-01-04:
If GNU Radio is built with ASAN (`-fsanitize=address`), it can be seen that most QT GUI sinks exhibit a use-after-free bug when the QT window is closed:
```
=================================================================
==3326439==ERROR: AddressSanitizer: heap-use-after-free on address 0x61500010a480 at pc 0x7f3a199d03c3 bp 0x7ffe3e46f410 sp 0x7ffe3e46f400
READ of size 8 at 0x61500010a480 thread T0
    #0 0x7f3a199d03c2 in gr::qtgui::time_sink_c_impl::~time_sink_c_impl() /home/argilo/prefix_311/src/gnuradio/gr-qtgui/lib/time_sink_c_impl.cc:90
    #1 0x7f3a199d0ba7 in virtual thunk to gr::qtgui::time_sink_c_impl::~time_sink_c_impl() (/home/argilo/prefix_311/lib/libgnuradio-qtgui.so.3.11.0git+0x2fd0ba7)
    #2 0x7f3a205935af in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() /usr/include/c++/11/bits/shared_ptr_base.h:168
    #3 0x7f3a205935af in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() /usr/include/c++/11/bits/shared_ptr_base.h:705
    #4 0x7f3a205935af in std::__shared_ptr<gr::qtgui::time_sink_c, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr() /usr/include/c++/11/bits/shared_ptr_base.h:1154
    #5 0x7f3a205935af in std::shared_ptr<gr::qtgui::time_sink_c>::~shared_ptr() /usr/include/c++/11/bits/shared_ptr.h:122
    #6 0x7f3a205935af in pybind11::class_<gr::qtgui::time_sink_c, gr::sync_block, gr::block, gr::basic_block, std::shared_ptr<gr::qtgui::time_sink_c> >::dealloc(pybind11::detail::value_and_holder&) /usr/include/pybind11/pybind11.h:1670
    #7 0x7f3a1ff71a4f in pybind11::detail::clear_instance(_object*) /usr/include/pybind11/detail/class.h:399
    #8 0x7f3a1ff77590 in pybind11_object_dealloc /usr/include/pybind11/detail/class.h:419
    #9 0x56189e9238b2 in _Py_Dealloc ../Objects/object.c:2301
    #10 0x56189e9238b2 in _Py_DECREF ../Include/object.h:500
    #11 0x56189e9238b2 in _Py_XDECREF ../Include/object.h:567
    #12 0x56189e9238b2 in dict_dealloc ../Objects/dictobject.c:2068
    #13 0x7f3a2ce66054  (/usr/lib/python3/dist-packages/PyQt5/sip.cpython-310-x86_64-linux-gnu.so+0x17054)
    #14 0x7f3a2ce66107  (/usr/lib/python3/dist-packages/PyQt5/sip.cpython-310-x86_64-linux-gnu.so+0x17107)
    #15 0x7f3a2ce6acbd  (/usr/lib/python3/dist-packages/PyQt5/sip.cpython-310-x86_64-linux-gnu.so+0x1bcbd)
    #16 0x56189e95fc65 in subtype_dealloc ../Objects/typeobject.c:1460
    #17 0x56189e9682d6 in _Py_Dealloc ../Objects/object.c:2301
    #18 0x56189e9682d6 in _Py_DECREF ../Include/object.h:500
    #19 0x56189e9682d6 in _Py_XDECREF ../Include/object.h:567
    #20 0x56189e9682d6 in cell_dealloc ../Objects/cellobject.c:82
    #21 0x56189e922c51 in _Py_Dealloc ../Objects/object.c:2301
    #22 0x56189e922c51 in _Py_DECREF ../Include/object.h:500
    #23 0x56189e922c51 in _Py_XDECREF ../Include/object.h:567
    #24 0x56189e922c51 in tupledealloc ../Objects/tupleobject.c:276
    #25 0x56189e944fc8 in _Py_Dealloc ../Objects/object.c:2301
    #26 0x56189e944fc8 in _Py_DECREF ../Include/object.h:500
    #27 0x56189e944fc8 in func_clear ../Objects/funcobject.c:636
    #28 0x56189e944fc8 in func_dealloc ../Objects/funcobject.c:648
    #29 0x56189ea4d5a3 in _Py_DECREF ../Include/object.h:500
    #30 0x56189ea4d5a3 in _Py_XDECREF ../Include/object.h:567
    #31 0x56189ea4d5a3 in _PySignal_Fini ../Modules/signalmodule.c:1759
    #32 0x56189ea4d5a3 in Py_FinalizeEx ../Python/pylifecycle.c:1767
    #33 0x56189ea3e912 in Py_RunMain ../Modules/main.c:668
    #34 0x56189ea1502c in Py_BytesMain ../Modules/main.c:720
    #35 0x7f3a2cc29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #36 0x7f3a2cc29e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #37 0x56189ea14f24 in _start (/usr/bin/python3.10+0x227f24)

0x61500010a480 is located 0 bytes inside of 472-byte region [0x61500010a480,0x61500010a658)
freed by thread T0 here:
    #0 0x7f3a2d4b724f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
    #1 0x7f3a290e4a6d in QObjectPrivate::deleteChildren() (/lib/x86_64-linux-gnu/libQt5Core.so.5+0x2e4a6d)

previously allocated by thread T0 here:
    #0 0x7f3a2d4b61e7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7f3a199c8d5d in gr::qtgui::time_sink_c_impl::initialize() /home/argilo/prefix_311/src/gnuradio/gr-qtgui/lib/time_sink_c_impl.cc:111

SUMMARY: AddressSanitizer: heap-use-after-free /home/argilo/prefix_311/src/gnuradio/gr-qtgui/lib/time_sink_c_impl.cc:90 in gr::qtgui::time_sink_c_impl::~time_sink_c_impl()
Shadow bytes around the buggy address:
  0x0c2a80019440: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a80019450: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a80019460: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a80019470: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a80019480: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c2a80019490:[fd]fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a800194a0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a800194b0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c2a800194c0: fd fd fd fd fd fd fd fd fd fd fd fa fa fa fa fa
  0x0c2a800194d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c2a800194e0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==3326439==ABORTING
```

The use-after-free occurs when the GNU Radio block's destructor checks whether it needs to close the corresponding QT widget:

https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gr-qtgui/lib/time_sink_c_impl.cc#L88-L92

This mechanism was added in 989ea0b76e5c4edddcd59e549bce1045536b144d, and it appears the goal was to allow GNU Radio blocks to close their associated QT widgets when they are removed from a flow graph. Unfortunately the mechanism doesn't work properly if the user closes the QT window, because doing so deletes all QT widgets, and it is not legal to call `d_main_gui->isClosed()` after this point.

The following QT GUI sinks have this bug:

* QT GUI Bercurve Sink
* QT GUI Constellation Sink
* QT GUI Eye Sink (complex, float)
* QT GUI Frequency Sink (complex, float)
* QT GUI Histogram Sink
* QT GUI Matrix Sink
* QT GUI Time Raster Sink (byte, float)
* QT GUI Time Sink (complex, float)
* QT GUI Vector Sink
* QT GUI Waterfall Sink (complex, float)

The remaining sinks do not have the bug, because the GNU Radio block doesn't attempt to close the QT widget:

* QT GUI Number Sink
* QT GUI Sink (complex, float)

Unfortunately I don't know Qt well enough to know what an appropriate fix would be. Perhaps this closing mechanism is not so important, if not all of the sinks use it. It's not clear from the commit message what problem it was meant to solve.

marcusmueller, 2024-01-08:
I think I ran into something similar before, in a different project. Either it was that Qt5's `::deleteChildren`is simply actually doing that, or we're not "using it correctly", though I really must say, it's not easy to figure out how to do that correctly. @mariusBgm, you wanted FOSS street cred?

argilo, 2024-01-08:
Perhaps referencing the `QWidget` using a `QPointer` instead of a raw pointer would solve the problem, since that will be automatically nulled when Qt deletes the `QWidget`.

https://doc.qt.io/qt-6/qpointer.html

argilo, 2024-01-08:
> Perhaps referencing the `QWidget` using a `QPointer` instead of a raw pointer would solve the problem

This seems to work well. I've opened #7065 with that change.

PR #7065 (2024-01-13): qtgui: Fix use-after-free in block destructors
## Description
As explained in #7050, most QT GUI sinks have a use-after-free bug in their destructors, which occurs when the user closes the GUI window. The Qt runtime deletes the the sink's `QWidget`, and then the sink attempts to dereference the deleted `QWidget` and call `isClosed()`.

To fix the problem, I've replaced the raw pointers with `QPointer`, which is automatically nulled when the underlying `QWidget` is deleted. ~~This provides a convenient way for the sink to check whether its `QWidget` still exists~~ The `QPointer` can then be use [... cut]
closed 2024-01-13 by willcode