gnuradio/gnuradio #7056: Failure to build 3.10.9.1 with `CXXFLAGS="-flto -Werror=odr"`

ZeroChaos-, 2024-01-05:
### What happened?

I tried to build with `CXXFLAGS="-flto -Werror=odr"` and it failed

### System Information

OS: Linux
GR Installation Method: Gentoo package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.1

### Steps to Reproduce the Problem

attempt to build with `CXXFLAGS="-flto -Werror=odr"`

### Relevant log output

```shell
[1021/1629] : && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -mtune=core2 -O3 -frecord-gcc-switches -pipe -flto -Werror=strict-aliasing -Werror=odr -Werror=lto-type-mismatch -Wstringop-overread -Werror=stringop-overread -march=native -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -msse2  -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libgnuradio-dtv.so.3.10.9 -o gr-dtv/lib/libgnuradio-dtv.so.3.10.9.1 gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_plinfo.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_deinterleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_depad_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_derandomizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_equalizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fpll_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fs_checker_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_randomize.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_rs_decoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_sync_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fake_single_viterbi.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_single_viterbi.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_viterbi_decoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_pad_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_randomizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_rs_encoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_interleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_basic_trellis_encoder.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_trellis_encoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_field_sync_mux_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bbheader_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bbscrambler_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bch_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_ldpc_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_interleaver_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_modulator_bc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_cellinterleaver_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_framemapper_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_freqinterleaver_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_pilotgenerator_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_paprtr_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_p1insertion_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_miso_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbs2/dvbs2_interleaver_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbs2/dvbs2_modulator_bc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbs2/dvbs2_physical_cc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_energy_dispersal_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_reed_solomon_enc_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_convolutional_interleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_configure.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_inner_coder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_bit_inner_interleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_symbol_inner_interleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_map_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_reference_signals_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_ofdm_sym_acquisition_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_demod_reference_signals_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_demap_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_bit_inner_deinterleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_viterbi_decoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_convolutional_deinterleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_reed_solomon_dec_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt/dvbt_energy_descramble_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/catv/catv_transport_framing_enc_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/catv/catv_reed_solomon_enc_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/catv/catv_randomizer_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/catv/catv_frame_sync_enc_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/catv/catv_trellis_enc_bb_impl.cc.o  -Wl,-rpath,/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gr-analog/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gr-filter/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gr-fec/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gr-fft/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gr-blocks/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gnuradio-runtime/lib:/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1_build/gnuradio-runtime/lib/pmt:  gr-analog/lib/libgnuradio-analog.so.3.10.9.1  gr-filter/lib/libgnuradio-filter.so.3.10.9.1  gr-fec/lib/libgnuradio-fec.so.3.10.9.1  gr-fft/lib/libgnuradio-fft.so.3.10.9.1  gr-blocks/lib/libgnuradio-blocks.so.3.10.9.1  gnuradio-runtime/lib/libgnuradio-runtime.so.3.10.9.1  gnuradio-runtime/lib/pmt/libgnuradio-pmt.so.3.10.9.1  /usr/lib64/libboost_thread.so.1.82.0  /usr/lib64/libspdlog.so.1.11.0  /usr/lib64/libfmt.so.9.1.0  /usr/lib64/libgmpxx.so  /usr/lib64/libgmp.so  -lrt  /usr/lib64/libvolk.so.3.1.0  -ldl  -lm && :
FAILED: gr-dtv/lib/libgnuradio-dtv.so.3.10.9.1 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -mtune=core2 -O3 -frecord-gcc-switches -pipe -flto -Werror=strict-aliasing -Werror=odr -Werror=lto-type-mismatch -Wstringop-overread -Werror=stringop-overread -march=native -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -fcx-limited-range -msse2  -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0 -shared -Wl,-soname,libgnuradio-dtv.so.3.10.9 -o gr-dtv/lib/libgnuradio-dtv.so.3.10.9.1 gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_plinfo.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_deinterleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_depad_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_derandomizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_equalizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fpll_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fs_checker_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_randomize.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_rs_decoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_sync_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_fake_single_viterbi.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_single_viterbi.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_viterbi_decoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_pad_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_randomizer_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_rs_encoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_interleaver_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_basic_trellis_encoder.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_trellis_encoder_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/atsc/atsc_field_sync_mux_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bbheader_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bbscrambler_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_bch_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvb/dvb_ldpc_bb_impl.cc.o gr-dtv/lib/CMakeFiles/gnuradio-dtv.dir/dvbt2/dvbt2_interleaver_bb_impl.cc [... cut ...]
/var/tmp/portage/net-wireless/gnuradio-3.10.9.1/work/gnuradio-3.10.9.1/gr-dtv/lib/dvbt2/dvbt2_paprtr_cc_impl.h:21:6: note: an enum with different number of values is defined in another translation unit
   21 | enum dvbt2_carrier_type_t {
      |      ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status

```
```

willcode, 2024-01-05:
The enum in pilotgenerator is a superset of the one in paprtr. I don't see both of them included in the same compilation unit, in which case this wouldn't actually be a problem. We'd also need to know whether all the items in pilotgenerator are applicable in paprtr. Not inclined to do anything about this in a hurry.

marcusmueller, 2024-01-08:
@willcode but a superset need not lead to the same underlying integer type. So, agreed, super unlikely that this is problematic, but we should probably still move the enum definition out to its own public header, or at least figure out why these two enums that seem to only be in impl headers end up as API that LTO tries to merge.

willcode, 2024-01-08:
I'm surprised that LTO cares about redefinitions where it doesn't matter. What if two people used the same name for completely unrelated types? Anyway, I agree that defining the single type somewhere and including it is better programming IF both blocks mean to deal with all the same set of values. Again, you can imagine that one block supports more values than the other because code didn't get updated or something.

argilo, 2024-01-09:
> I'm surprised that LTO cares about redefinitions where it doesn't matter.

It looks like it does matter for the C++ One Definition Rule, which says that if a program contains more than one definition of a type, then each definition must be equivalent.

willcode, 2024-01-09:
Renaming one of the enums, or using anonymous enums, should be sufficient to get rid of the error, if no other analysis is done. The enum types are never used by name. Interaction between the two blocks (whether they should support the same carrier types) is more interesting.