gnuradio/gnuradio #7059: Regression: GRC fails to compile flow graphs with bus connections

argilo, 2024-01-06:
In GNU Radio 3.10.9.1, GRC fails to compile the following flowgraph:

[image]

The Null Source has two complex outputs, and the Null Sink has two complex inputs.

A bisect shows that the problem began with https://github.com/gnuradio/gnuradio/pull/6889.

argilo, 2024-01-06:
Strangely, if bus sizes are mismatched, then the flow graph compiles without issue.

willcode, 2024-01-06:
Similar problem with virtual sink/source was fixed in #7016

mbr0wn, 2024-01-07:
I think I know how to fix this, but I can't replicate the original issue. I'm having issues with all examples that have bus ports, though.

PR #7067 (2024-01-08): grc: Fix compilation of bus ports
This removes a superfluous call to `Connection.__init__()` in the top block generator which uses itself as a parent, instead of the original flowgraph as elsewhere in the same file. 0fd504e5 changed the Connection object such that passing in a parent that is not an Element is no longer valid.

## Description

## Related Issue

Fixes #7059 

## Which blocks/areas does this affect?

GRC + Bus Ports

## Testing Done

Used the same FG from #7059 to test both crash and fix. Also, just by looking at the code, this seems like a safe change as the `connection` object  [... cut]
closed 2024-01-08 by willcode