gnuradio/gnuradio #7064: GRC: Revise Options Block YML documentation section

duggabe, 2024-01-07:
### What happened?

The documentation section of the options.block.yml file needs to be updated to reflect the current parameters and their usage.
All of the 'Bokeh GUI only' items should be grouped together and marked as such.
Also it would be helpful to group the Parameters in the same order.

### System Information

OS: Ubuntu 22.04.3
GR Installation Method: Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0

### Steps to Reproduce the Problem

With GRC, select the Options block Properties -> Documentation tab

haakov, 2024-01-10:
Would it make sense to group all the GUI-related parameters in a separate tab in the Properties dialog?

duggabe, 2024-01-13:
@haakov yes, that would be good.

vamsip-cell, 2026-10-11:
Hi @duggabe! I do not see any active PRs for this issue, so I would like to pick this up. I will work on updating the Options block documentation and open a PR soon.