gnuradio/gnuradio #7075: Tag propagation in Integrator is broken

daniestevez, 2024-01-09:
### What happened?

Tag propagation in the Integrator block is not working as expected. The following flowgraph demonstrates the problem.

[test_integrate.grc.zip](https://github.com/gnuradio/gnuradio/files/13876385/test_integrate.grc.zip)

[image]

24 samples all having the value `1.0` are generated, each having a tag whose value is a counter (0, 1, 2, 3, ...). These are integrated in blocks of 6 to produce 4 outputs. The output tags of the Integrator are printed with a Tag Debug block, and when the flowgraph finishes the values of the output samples are also printed. This is what the flowgraph prints.

```
----------------------------------------------------------------------
Tag Debug: 
Input Stream: 00
  Offset: 0  Source: n/a     Key: key   Value: 0
  Offset: 0  Source: n/a     Key: key   Value: 1
  Offset: 0  Source: n/a     Key: key   Value: 2
  Offset: 1  Source: n/a     Key: key   Value: 3
  Offset: 1  Source: n/a     Key: key   Value: 4
  Offset: 1  Source: n/a     Key: key   Value: 5
  Offset: 1  Source: n/a     Key: key   Value: 6
  Offset: 1  Source: n/a     Key: key   Value: 7
  Offset: 1  Source: n/a     Key: key   Value: 8
  Offset: 2  Source: n/a     Key: key   Value: 9
  Offset: 2  Source: n/a     Key: key   Value: 10
  Offset: 2  Source: n/a     Key: key   Value: 11
  Offset: 2  Source: n/a     Key: key   Value: 12
  Offset: 2  Source: n/a     Key: key   Value: 13
  Offset: 2  Source: n/a     Key: key   Value: 14
  Offset: 3  Source: n/a     Key: key   Value: 15
  Offset: 3  Source: n/a     Key: key   Value: 16
  Offset: 3  Source: n/a     Key: key   Value: 17
  Offset: 3  Source: n/a     Key: key   Value: 18
  Offset: 3  Source: n/a     Key: key   Value: 19
  Offset: 3  Source: n/a     Key: key   Value: 20
----------------------------------------------------------------------
[6.0, 6.0, 6.0, 6.0]
```

The values of all the output samples are `6.0`, meaning that the integrator has correctly added together 6 samples always. However, the first output only carries 3 tags with values 0, 1, 2, instead of 6 tags with values 0, 1, 2, 3, 4, 5. The next output carries 6 tags, but these are 3 through 8 rather than 6 through 11, etc.

### System Information

OS: Arch Linux
GR Installation Method: distro package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0

### Steps to Reproduce the Problem

Run the flowgraph linked in this issue.

eltos, 2024-05-08:
Looks like this is related to the rounding of offsets in the tag propagation implementation? I.e. the offset of tag with value 3 is computed as 3/6=0.5 and then rounded to 1.

Should offsets be in general rounded down maybe?

Possibly related to #7328

daniestevez, 2024-05-09:
> The offset of tag with value 3 is computed as 3/6=0.5 and then rounded to 1.

That was my guess as to why this happens.

> Should offsets be in general rounded down maybe?

In this particular case I think yes. In general, I'm not sure. Perhaps for a Lowpass FIR Filter block doing decimation I would expect tags to appear in the nearest output sample (round rather than floor).

marcusmueller, 2024-05-13:
I'd think that for any tag with a fractional offset N.y, we need to put it on item N, as otherwise it might not get transported (e.g., when the `work` calls get split between the Nth and (N+1)th item). So, rounding down the offset seems correct, in most cases; but quite honestly, I'm not too sure – this is really a bit of a situation where we try to force the concept of precise timing onto a sampling system.

Anyways, yes, losing the tag here is more than unfortunate, and if rounding down solves it here, that might be the right thing to do, just for not-losing-tags reasons, no matter whether we are doing the correct thing, position-wise.

daniestevez, 2024-05-14:
The integrator is a `sync_decimator` block. Doesn't that mean that in each call to `work()` the number of input items is a multiple of `decim` and `nintput_items / decim` are produced exactly from those input items? I don't see that there is a chance for a `work()` call to get split as you say.

marcusmueller, 2024-05-14:
>  I don't see that there is a chance for a work() call to get split as you say.

I had this scenario in mind:

Say decim==4, and ninput_items==1024; -> noutput= 256. First call to `integrator::work`, so nitems_read(0) == 0.

Say there's a tag on the 1023. input item (so, last input item), then block_executor will, in `propagate_tags`, take the old offset ($1023$), multiply it with the relative rate ($\frac{1}{4}$), add $\frac{1}{2}$, and round down to get the offset of the new tag:

$$
o_{\text{new}} = \left\lfloor o_{\text{old}}\cdot r + \frac{1}{2} \right\rfloor =  \left\lfloor 1023\cdot \frac{1}{4} + \frac{2}{4}\right\rfloor =  \left\lfloor 256.25  \right\rfloor =  256
$$

Problem is that the index of the last item leaving integrator in that call is $255 < 256$. So, we're "tagging into the future". Good thing is that nothing checks bounds on tags when inserting, and `prune_tags` only prunes the past, not the future.

I can't really put my finger on where these tags go missing.

marcusmueller, 2024-05-15:
yep, these "tagging rules" is also what I've been following. And I was really confused to see that it wasn't the last item that was missing a tag (which could have been explained by the in-future tagging). Hm.

eltos, 2024-05-15:
@marcusmueller

> I can't really put my finger on where these tags go missing.

But you just explained it: The tags with values 21, 22 and 23 are missing as they got assigned to the (future) sample with offset 4 because of rounding. Due to the Head block in the flow graph and the fact that Tag Debug block uses `get_tags_in_range` corresponding to the range of samples it's working on, they are simply not printed.

> And I was really confused to see that it wasn't the last item that was missing a tag

But it is, the tags on sample 0 are not missing but unexpectedly assigned to sample 1, and so on until the last sample.