gnuradio/gnuradio #7085: filter_qa_fir_filter_with_buffer.cc_test.sh failure

ZeroChaos-, 2024-01-18:
### What happened?

A test failed

### System Information

OS: Gentoo Linux
GR Installation Method: distro package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

build under gentoo and run tests, probably

### Relevant log output

```shell
----------------------------------------------------------

131/265 Testing: filter_qa_fir_filter_with_buffer.cc
131/265 Test: filter_qa_fir_filter_with_buffer.cc
Command: "/bin/sh" "filter_qa_fir_filter_with_buffer.cc_test.sh"
Directory: /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2_build/gr-filter/lib
"filter_qa_fir_filter_with_buffer.cc" start time: Jan 17 22:03 EST
Output:
----------------------------------------------------------
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Running 9 test cases...
unknown location(0): fatal error: in "t1_fff": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(132): last checkpoint: "t1_fff" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t2_fff": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(134): last checkpoint: "t2_fff" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t3_fff": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(136): last checkpoint: "t3_fff" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::complex<float
>; _Alloc = volk::alloc<std::complex<float> >; reference = std::complex<float>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t1_ccc": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(223): last checkpoint: "t1_ccc" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::complex<float
>; _Alloc = volk::alloc<std::complex<float> >; reference = std::complex<float>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t2_ccc": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(225): last checkpoint: "t2_ccc" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::complex<float
>; _Alloc = volk::alloc<std::complex<float> >; reference = std::complex<float>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t3_ccc": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(227): last checkpoint: "t3_ccc" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t1_ccf": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(312): last checkpoint: "t1_ccf" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t2_ccf": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(314): last checkpoint: "t2_ccf" test entry
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = vo
lk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
unknown location(0): fatal error: in "t3_ccf": signal: SIGABRT (application abort requested)
/var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc(316): last checkpoint: "t3_ccf" test entry

*** 9 failures are detected in the test module "Master Test Suite"

<end of output>
Test time =   0.04 sec
----------------------------------------------------------
Test Failed.
"filter_qa_fir_filter_with_buffer.cc" end time: Jan 17 22:03 EST
"filter_qa_fir_filter_with_buffer.cc" time elapsed: 00:00:00
----------------------------------------------------------
```

willcode, 2024-01-18:
`gr-filter/lib/qa_fir_filter_with_buffer.cc` includes
```
...
    volk::vector<o_type> expected_output(OUTPUT_LEN);
    volk::vector<o_type> actual_output(OUTPUT_LEN);
...
        for (int ol = 0; ol <= OUTPUT_LEN; ol++) {
...
```
The `<=` looks like it's asking for trouble.

argilo, 2024-01-19:
Are you able to execute `gr-filter/lib/filter_qa_fir_filter_with_buffer.cc` (from the build directory) inside gdb, ideally with debug symbols? That might help track down exactly where the assertion failure occurs.

argilo, 2024-01-19:
> The <= looks like it's asking for trouble.

That part looks OK to me. It's just testing vector lengths up to and including the length of the buffer itself.

So far I haven't been able to reproduce any buffer overruns with valgrind or ASAN.

ZeroChaos-, 2024-01-23:
```
Starting program: /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2_build/gr-filter/lib/filter_qa_fir_filter_with_buffer.cc 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Running 9 test cases...
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = volk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff6ec57d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff6e75236 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff6e5d8b7 in __GI_abort () at abort.c:79
#4  0x00007ffff70e509f in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>)
    at /var/tmp/portage/sys-devel/gcc-13.2.1_p20230826/work/gcc-13-20230826/libstdc++-v3/src/c++11/debug.cc:61
#5  0x00007ffff7ebb9d5 in std::vector<float, volk::alloc<float> >::operator[] (this=0x555555580438, __n=7) at /usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125
#6  0x00007ffff7eccb50 in gr::filter::kernel::fir_filter_with_buffer_fff::set_taps (this=0x7fffffffc210, taps=std::vector of length 0, capacity 0)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/fir_filter_with_buffer.cc:55
#7  0x00007ffff7ecc951 in gr::filter::kernel::fir_filter_with_buffer_fff::fir_filter_with_buffer_fff (this=0x7fffffffc210, taps=std::vector of length 0, capacity 0)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/fir_filter_with_buffer.cc:32
#8  0x000055555555ac64 in gr::filter::fff::test_decimate (decimate=1)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:111
#9  0x000055555555b88a in gr::filter::fff::t1_fff::test_method (this=0x7fffffffc2fe)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:132
#10 0x000055555555b4c0 in gr::filter::fff::t1_fff_invoker () at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:132
#11 0x00005555555653f9 in boost::detail::function::void_function_invoker0<void (*)(), void>::invoke (function_ptr=...) at /usr/include/boost/function/function_template.hpp:117
#12 0x00007ffff7d51071 in boost::detail::function::function_obj_invoker0<boost::detail::forward, int>::invoke(boost::detail::function::function_buffer&) ()
   from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#13 0x00007ffff7d4fdb0 in boost::execution_monitor::catch_signals(boost::function<int ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#14 0x00007ffff7d4fe98 in boost::execution_monitor::execute(boost::function<int ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#15 0x00007ffff7d50650 in boost::execution_monitor::vexecute(boost::function<void ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#16 0x00007ffff7d67289 in boost::unit_test::unit_test_monitor_t::execute_and_translate(boost::function<void ()> const&, unsigned long) ()
   from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#17 0x00007ffff7d5451a in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned long, boost::unit_test::framework::state::random_generator_helper const*) [clone .isra.0] () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#18 0x00007ffff7d53e79 in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned long, boost::unit_test::framework::state::random_generator_helper const*) [clone .isra.0] () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#19 0x00007ffff7d56489 in boost::unit_test::framework::run(unsigned long, bool) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#20 0x00007ffff7d6671a in boost::unit_test::unit_test_main(bool (*)(), int, char**) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#21 0x000055555555a704 in main (argc=1, argv=0x7fffffffd698) at /usr/include/boost/test/unit_test.hpp:64
```

argilo, 2024-01-23:
OK, I see what's happening. `qa_fir_filter_with_buffer` tests filters with 0 to 29 taps:

https://github.com/gnuradio/gnuradio/blob/ff61502dcca8bdc1ede3c8dfa113635c1983a2c0/gr-filter/lib/qa_fir_filter_with_buffer.cc#L90

When there are zero taps, `fir_filter_with_buffer` resizes its aligned buffers to length `0 + d_naligned - 1`, and then on line 55 accesses index `i`, with `i` ranging from `0` to `d_naligned - 1`. The last access is out of bounds:

https://github.com/gnuradio/gnuradio/blob/ff61502dcca8bdc1ede3c8dfa113635c1983a2c0/gr-filter/lib/fir_filter_with_buffer.cc#L51-L56

ZeroChaos-, 2024-01-23:
I'm positive I applied the patch, but it looks similar to me
```
(gdb) run
Starting program: /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2_build/gr-filter/lib/filter_qa_fir_filter_with_buffer.cc 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Running 9 test cases...
/usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = float; _Alloc = volk::alloc<float>; reference = float&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff6ec57d3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff6e75236 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff6e5d8b7 in __GI_abort () at abort.c:79
#4  0x00007ffff70e509f in std::__glibcxx_assert_fail (file=<optimized out>, line=<optimized out>, function=<optimized out>, condition=<optimized out>)
    at /var/tmp/portage/sys-devel/gcc-13.2.1_p20230826/work/gcc-13-20230826/libstdc++-v3/src/c++11/debug.cc:61
#5  0x0000555555561d2f in std::vector<float, volk::alloc<float> >::operator[] (this=0x7fffffffc190, __n=29) at /usr/lib/gcc/x86_64-pc-linux-gnu/13/include/g++-v13/bits/stl_vector.h:1125
#6  0x000055555555ac05 in gr::filter::fff::test_decimate (decimate=1)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:110
#7  0x000055555555b88a in gr::filter::fff::t1_fff::test_method (this=0x7fffffffc2fe)
    at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:132
#8  0x000055555555b4c0 in gr::filter::fff::t1_fff_invoker () at /var/tmp/portage/net-wireless/gnuradio-3.10.9.2/work/gnuradio-3.10.9.2/gr-filter/lib/qa_fir_filter_with_buffer.cc:132
#9  0x00005555555653f9 in boost::detail::function::void_function_invoker0<void (*)(), void>::invoke (function_ptr=...) at /usr/include/boost/function/function_template.hpp:117
#10 0x00007ffff7d51071 in boost::detail::function::function_obj_invoker0<boost::detail::forward, int>::invoke(boost::detail::function::function_buffer&) ()
   from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#11 0x00007ffff7d4fdb0 in boost::execution_monitor::catch_signals(boost::function<int ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#12 0x00007ffff7d4fe98 in boost::execution_monitor::execute(boost::function<int ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#13 0x00007ffff7d50650 in boost::execution_monitor::vexecute(boost::function<void ()> const&) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#14 0x00007ffff7d67289 in boost::unit_test::unit_test_monitor_t::execute_and_translate(boost::function<void ()> const&, unsigned long) ()
   from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#15 0x00007ffff7d5451a in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned long, boost::unit_test::framework::state::random_generator_helper const*) [clone .isra.0] () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#16 0x00007ffff7d53e79 in boost::unit_test::framework::state::execute_test_tree(unsigned long, unsigned long, boost::unit_test::framework::state::random_generator_helper const*) [clone .isra.0] () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#17 0x00007ffff7d56489 in boost::unit_test::framework::run(unsigned long, bool) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#18 0x00007ffff7d6671a in boost::unit_test::unit_test_main(bool (*)(), int, char**) () from /usr/lib64/libboost_unit_test_framework.so.1.82.0
#19 0x000055555555a704 in main (argc=1, argv=0x7fffffffd698) at /usr/include/boost/test/unit_test.hpp:64
```

argilo, 2024-01-23:
Actually the crash is different now. Let me see what's failing this time.

argilo, 2024-01-23:
Looks like the test itself also has Undefined Behaviour. Can you try this patch?
```diff
diff --git a/gr-filter/lib/qa_fir_filter_with_buffer.cc b/gr-filter/lib/qa_fir_filter_with_buffer.cc
index 32f5c6780f..72cdd9fa1c 100644
--- a/gr-filter/lib/qa_fir_filter_with_buffer.cc
+++ b/gr-filter/lib/qa_fir_filter_with_buffer.cc
@@ -107,7 +107,7 @@ void test_decimate(unsigned int decimate)
             }
 
             // build filter
-            vector<tap_type> f1_taps(&taps[0], &taps[n]);
+            vector<tap_type> f1_taps(taps.begin(), taps.begin() + n);
             kernel::fir_filter_with_buffer_fff f1(f1_taps);
 
             // zero the output, then do the filtering
@@ -198,7 +198,7 @@ void test_decimate(unsigned int decimate)
             }
 
             // build filter
-            vector<tap_type> f1_taps(&taps[0], &taps[n]);
+            vector<tap_type> f1_taps(taps.begin(), taps.begin() + n);
             kernel::fir_filter_with_buffer_ccc f1(f1_taps);
 
             // zero the output, then do the filtering
@@ -287,7 +287,7 @@ void test_decimate(unsigned int decimate)
             }
 
             // build filter
-            vector<tap_type> f1_taps(&taps[0], &taps[n]);
+            vector<tap_type> f1_taps(taps.begin(), taps.begin() + n);
             kernel::fir_filter_with_buffer_ccf f1(f1_taps);
 
             // zero the output, then do the filtering
```

argilo, 2024-01-23:
While you're at it, could you also test this alternate patch for the first bug, which avoids the need for an extra length check?

```diff
diff --git a/gr-filter/lib/fir_filter_with_buffer.cc b/gr-filter/lib/fir_filter_with_buffer.cc
index 4fc571ec57..ecd989b8aa 100644
--- a/gr-filter/lib/fir_filter_with_buffer.cc
+++ b/gr-filter/lib/fir_filter_with_buffer.cc
@@ -52,7 +52,7 @@ void fir_filter_with_buffer_fff::set_taps(const std::vector<float>& taps)
     for (int i = 0; i < d_naligned; i++) {
         d_aligned_taps[i].clear();
         d_aligned_taps[i].resize(d_ntaps + d_naligned - 1);
-        std::copy(std::begin(d_taps), std::end(d_taps), &d_aligned_taps[i][i]);
+        std::copy(std::begin(d_taps), std::end(d_taps), std::begin(d_aligned_taps[i]) + i);
     }
 
     d_idx = 0;
@@ -157,7 +157,7 @@ void fir_filter_with_buffer_ccc::set_taps(const std::vector<gr_complex>& taps)
     for (int i = 0; i < d_naligned; i++) {
         d_aligned_taps[i].clear();
         d_aligned_taps[i].resize(d_ntaps + d_naligned - 1);
-        std::copy(std::begin(d_taps), std::end(d_taps), &d_aligned_taps[i][i]);
+        std::copy(std::begin(d_taps), std::end(d_taps), std::begin(d_aligned_taps[i]) + i);
     }
 
     d_idx = 0;
@@ -261,7 +261,7 @@ void fir_filter_with_buffer_ccf::set_taps(const std::vector<float>& taps)
     for (int i = 0; i < d_naligned; i++) {
         d_aligned_taps[i].clear();
         d_aligned_taps[i].resize(d_ntaps + d_naligned - 1);
-        std::copy(std::begin(d_taps), std::end(d_taps), &d_aligned_taps[i][i]);
+        std::copy(std::begin(d_taps), std::end(d_taps), std::begin(d_aligned_taps[i]) + i);
     }
 
     d_idx = 0;
```

ZeroChaos-, 2024-01-23:
So test patch 2 and 3 or just 3?

argilo, 2024-01-23:
2 and 3

argilo, 2024-01-23:
I would guess these errors are visible because the `_GLIBCXX_ASSERTIONS` macro is set. I'll see if I can reproduce by building GNU Radio that way. It might be a good idea to do so in CI to catch more bugs.

ZeroChaos-, 2024-01-23:
Joy of joys
```
100% tests passed, 0 tests failed out of 265
```

argilo, 2024-01-23:
Glad to hear it! I'll open up a PR.