gnuradio/gnuradio #7086: light theme when using dark theme

marcusmueller, 2024-01-18:
### What happened?

GRC shows up in bright colors, though the dark theme is selected by default

[image]

Changes to dark them after switching to light, restarting grc, switching to dark, restarting GRC. 

### System Information

OS: fedora 39
GR Installation Method:

```shell
git clone -b feature-grc-qt https://github.com/gnuradio/gnuradio
cd gnuradio
python3 -m venv --system-site-packages .venv
source .venv/bin/activate
pip install QDarkStyle qtpy
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV ..
make -j8
make install
```

```
source /path/to/gnuradio/.venv/bin/activate
export LD_LIBRARY_PATH=$VIRTUAL_ENV/lib/
export GRC_BLOCKS_PATH=$VIRTUAL_ENV/share/gnuradio/grc/blocks/
export PYTHONPATH=$VIRTUAL_ENV/lib/python3.11/site-packages

gnuradio-companion --qt
```

### GNU Radio Version

QT-based Companion branch (feature-grc-qt)

### Specific Version

GNU Radio v3.11.0.0git-625-g60817bdf (Python 3.11.7)

### Steps to Reproduce the Problem

start GRC

marcusmueller, 2024-01-18:
~/.gnuradio/grc.log does not contain "Did not find QDarkstyle"

haakov, 2024-01-19:
Does https://github.com/gnuradio/gnuradio/commit/c98d9243c9422325bd25c6d3b549aa534e505456 (current feature-grc-qt branch) fix the issue?

luzpaz, 2024-02-14:
@marcusmueller :point_up_2:

willcode, 2024-02-14:
@luzpaz are you saying it works for you?

luzpaz, 2024-02-14:
I hadn't been able to reproduce on Manjaro. So i was wondering if it was a fedora related issues

closed 2024-03-11 by marcusmueller