gnuradio/gnuradio #7087: QT-based companion crashes with Python 3.9 on openSUSE 15.5

hennichodernich, 2024-01-20:
### What happened?

GNU Radio companion crashes when you try to generate the flowgraph.

### System Information

OS: openSUSE Leap 15.5
Python 3.9 from openSUSE backports repo (because stock Python is 3.6), all required modules installed via pip3.9
GR Installation Method: compiled from source using these instructions: 
https://mastodon.social/@funkylab/111776932485011379
all modules compiled except gr-uhd

### GNU Radio Version

QT-based Companion branch (feature-grc-qt)

### Specific Version

v3.11.0.0git-625-g60817bdf

### Steps to Reproduce the Problem

Try to generate any flowgraph.
[image]

### Relevant log output

```shell
(.venv) henning@watson:~/src/gnuradio/build> gnuradio-companion --qt
Handler not implemented for <vertical_align_top.triggered> in MainWindow
Handler not implemented for <vertical_align_middle.triggered> in MainWindow
Handler not implemented for <vertical_align_bottom.triggered> in MainWindow
Handler not implemented for <horizontal_align_left.triggered> in MainWindow
Handler not implemented for <horizontal_align_center.triggered> in MainWindow
Handler not implemented for <horizontal_align_right.triggered> in MainWindow
Handler not implemented for <create_hier.triggered> in MainWindow
Handler not implemented for <open_hier.triggered> in MainWindow
Handler not implemented for <toggle_grid.triggered> in MainWindow
Handler not implemented for <set_default_qt_gui_theme.triggered> in MainWindow
Handler not implemented for <reload.triggered> in MainWindow
Could not restore window geometry and state.
Could not connect <auto_scroll.toggled> to handler <auto_scroll_toggled>
Handler not implemented for <auto_scroll.triggered> in Console
Traceback (most recent call last):
  File "/home/henning/src/gnuradio/.venv/lib/python3.9/site-packages/gnuradio/grc/gui_qt/components/window.py", line 1231, in generate_triggered
    self.save_triggered()
  File "/home/henning/src/gnuradio/.venv/lib/python3.9/site-packages/gnuradio/grc/gui_qt/components/window.py", line 937, in save_triggered
    self.platform.save_flow_graph(filename, self.currentView)
  File "/home/henning/src/gnuradio/.venv/lib/python3.9/site-packages/gnuradio/grc/core/platform.py", line 367, in save_flow_graph
    data = flow_graph.export_data()
AttributeError: 'FlowgraphView' object has no attribute 'export_data'
Abgebrochen (Speicherabzug geschrieben)
```

hennichodernich, 2024-01-20:
Addendum: Starting without the --qt parameter makes the companion work flawlessly.

List of installed python modules:
henning@watson:~> /usr/local/bin/pip3.9  list
Package                   Version
------------------------- ---------
attrs                     23.2.0
click                     8.1.7
click-plugins             1.1.1
importlib-metadata        7.0.1
jsonschema                4.21.0
jsonschema-specifications 2023.12.1
Mako                      1.3.0
MarkupSafe                2.1.3
numpy                     1.26.3
packaging                 23.2
pip                       23.3.2
pyaml                     23.12.0
pybind11                  2.11.1
pycairo                   1.25.1
pygccxml                  2.4.0
PyGObject                 3.46.0
PyQt5                     5.15.10
PyQt5-Qt5                 5.15.2
PyQt5-sip                 12.13.0
PyYAML                    6.0.1
QDarkStyle                3.2.3
QtPy                      2.4.1
referencing               0.32.1
rpds-py                   0.17.1
scipy                     1.11.4
setuptools                44.1.1
sip                       6.8.1
six                       1.16.0
tomli                     2.0.1
zipp                      3.17.0

haakov, 2024-01-21:
Thank you for the report! The problem should be fixed with https://github.com/gnuradio/gnuradio/commit/92b1f07cd3b84bead9f63f1ca56beeccb8b15e83, can you confirm that it works if you do another git pull?

hennichodernich, 2024-01-21:
Yes, this fixes this particular issue.

closed 2024-01-21 by hennichodernich