gnuradio/gnuradio #7089: Can't run generated flowgraph due to print formatting

hennichodernich, 2024-01-21:
### What happened?

The generated flowgraph doesn't run but causes a syntax error in the Python interpreter.

### System Information

OS: openSUSE Leap 15.5
Python 3.9 from openSUSE backports repo (because stock Python is 3.6), all required modules installed via pip3.9
GR Installation Method: compiled from source using these instructions:
https://mastodon.social/@funkylab/111776932485011379
all modules compiled except gr-uhd

### GNU Radio Version

QT-based Companion branch (feature-grc-qt)

### Specific Version

v3.11.0.0git-636-g985cfd06

### Steps to Reproduce the Problem

Click "Execute" or press F6 in companion

### Relevant log output

```shell
(.venv) henning@watson:~/src/gnuradio/build> gnuradio-companion --qt
Handler not implemented for <vertical_align_top.triggered> in MainWindow
Handler not implemented for <vertical_align_middle.triggered> in MainWindow
Handler not implemented for <vertical_align_bottom.triggered> in MainWindow
Handler not implemented for <horizontal_align_left.triggered> in MainWindow
Handler not implemented for <horizontal_align_center.triggered> in MainWindow
Handler not implemented for <horizontal_align_right.triggered> in MainWindow
Handler not implemented for <create_hier.triggered> in MainWindow
Handler not implemented for <open_hier.triggered> in MainWindow
Handler not implemented for <toggle_grid.triggered> in MainWindow
Handler not implemented for <set_default_qt_gui_theme.triggered> in MainWindow
Handler not implemented for <reload.triggered> in MainWindow
Could not restore window geometry and state.
Could not connect <auto_scroll.toggled> to handler <auto_scroll_toggled>
Handler not implemented for <auto_scroll.triggered> in Console
>>> Warning: The block 'blocks_throttle_0' is deprecated.
  File "/home/henning/gnuradio/noisedemo.py", line 40
    print(f"Qt GUI: Could not set Icon: {str(exc)}", file=sys.stderr)
                                                  ^
SyntaxError: invalid syntax
```

willcode, 2024-01-21:
Is this somehow running Python2? The try/except from that code runs fine on Ubuntu20.04 with Python 3.8.10.

haakov, 2024-01-22:
Thank you for the report! I've got a fix 90% ready for this issue.

I'll push it out this afternoon

hennichodernich, 2024-01-22:
Please note that it works flawlessly if I call the script from outside
the companion using just "python /home/henning/gnuradio/noisedemo.py".
It seems like the wrong python interpreter is chosen by the companion.

haakov, 2024-02-02:
@hennichodernich Sorry about the delay. You should notice an improvement if you pull the current `feature-grc-qt` branch.

hennichodernich, 2024-02-03:
Yes, the bug is gone, the flowgraph can successfully be started from within the companion.

closed 2024-02-03 by hennichodernich