gnuradio/gnuradio #7109: gr_modtool generated CMakeList.txt didn't include python file when adding a python source

thomascatlee, 2024-02-02:
### What happened?

gr_modtool add -t sync -l python

the python file will not be installed into gnuradio, by reason of python filename absence in CMakeList.txt gr_python_install section.

### System Information

OS: Win11
Installed radioconda-2023.11.06-Windows-x86_64.exe

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.8.0

### Steps to Reproduce the Problem

(base) D:\work\gnuradio>gr_modtool newmod tutorial
Creating out-of-tree module in .\gr-tutorial...
Done.
Use 'gr_modtool add' to add a new block to this currently empty module.

(base) D:\work\gnuradio>cd gr-tutorial

(base) D:\work\gnuradio\gr-tutorial>gr_modtool add -t sync -l python
GNU Radio module name identified: tutorial
Enter name of block/code (without module name prefix): multiply_py_ff
Language: Python
Block/code identifier: multiply_py_ff
Please specify the copyright holder:
Enter valid argument list, including default arguments:
 multiple
Add Python QA code? [Y/n] y
Adding file 'python\tutorial\multiply_py_ff.py'...
Adding file 'python\tutorial\qa_multiply_py_ff.py'...
Editing python\tutorial/CMakeLists.txt...
Adding file 'grc\tutorial_multiply_py_ff.block.yml'...
Editing grc/CMakeLists.txt...

---------------------------------------------------
python\tutorial/CMakeList.txt 
---------------------------------------------------
# Copyright 2011 Free Software Foundation, Inc.
#
# This file was generated by gr_modtool, a tool from the GNU Radio framework
# This file is a part of gr-tutorial
#
# SPDX-License-Identifier: GPL-3.0-or-later
#

########################################################################
# Include python install macros
########################################################################
include(GrPython)
if(NOT PYTHONINTERP_FOUND)
    return()
endif()

add_subdirectory(bindings)

########################################################################
# Install python sources
########################################################################
gr_python_install(FILES __init__.py DESTINATION ${GR_PYTHON_DIR}/gnuradio/tutorial)

########################################################################
# Handle the unit tests
########################################################################
include(GrTest)

set(GR_TEST_TARGET_DEPS gnuradio-tutorial)

# Create a package directory that tests can import. It includes everything
# from `python/`.
add_custom_target(
    copy_module_for_tests ALL
    COMMAND ${CMAKE_COMMAND} -E copy_directory ${CMAKE_CURRENT_SOURCE_DIR}
            ${PROJECT_BINARY_DIR}/test_modules/gnuradio/tutorial/)
GR_ADD_TEST(qa_multiply_py_ff ${PYTHON_EXECUTABLE} ${CMAKE_CURRENT_SOURCE_DIR}/qa_multiply_py_ff.py)
-----------------------------------------------------------------------------------------------
python\tutorial\multiply_py_ff.py will not be installed into gnuradio directory, so when you "from gnuradio import tutorial" you will get an error "ImportError: cannot import name 'tutorial' from 'gnuradio' "

willcode, 2024-02-02:
This sounds like it could be related to #7009, which was just merged on `main`.

marcusmueller, 2024-03-11:
No reaction in a month to a claimed duplicity - closing. But please reopen if wrongly closed!

closed 2024-03-11 by marcusmueller