gnuradio/gnuradio #7113: CtrlPort monitor crash on click

Notou, 2024-02-06:
### What happened?

When clicking on a displayed value, `gr-ctrlport-monitor` crashes.

### System Information

OS: Linux Mint 21.3
GR Installation Method: Conda 

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Run a flowgraph with CtrlPorts enabled
Connect to the running flowgraph with `gr-ctrlport-monitor <PORT>`
Click on one of the lines of the displayed table (doesn't matter which)
Crash

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/cyrille/conda/envs/radioconda/bin/gr-ctrlport-monitor", line 210, in startDrag
    iscomplex = (disp & gr.DISPOPTCPLX) or (disp & gr.DISPXY)
                        ^^^^^^^^^^^^^^
AttributeError: module 'gnuradio.gr' has no attribute 'DISPOPTCPLX'
Abandon (core dumped)
```

Notou, 2024-02-06:
I did some investigation:
The `DISPOPTCPLX` constant, as well as `DISPNULL`, `DISPTIME`, ... 
were not exposed to python via pybind11.

So I tried to add that to rpccallbackregister_base_python.cc, and it seems to solve that specific issue, and allow to see time (and PSD) plots of the ctrlport values.

I have that ready to PR.
But that is leading me into a rabbit hole of other ctrlport non working elements:
- Complex values that cause a crash when displayed (I thing I have a fix for that)
- The CtrlPort Monitor and CtrlPort Perfomance Monitor in GRC that don't start their related tool (Only the cmdline version works right now)
- The CtrlPort Probes don't seem to be exposed to python
And possibly more if I keep digging.

Should I make individual Issues and/or PRs for each?

willcode, 2024-02-06:
Thanks for looking into this.

A single PR for all CtrlPort fixes would be good. You can use multiple commits in the PR if you think the might be applied/reverted/bisected separately.

Notou, 2024-02-07:
Ok, I've drafted a PR with the fixes I have for now.

PR #7119 (2024-02-22): Add/Restore ControlPorts bindings

## Description

Adds and restore pybind11 bindings with ControlPorts related elements

The CtrlPort Monitor and CtrlPort Perfomance Monitor blocks in GRC don't currently work.
These seem to require some more pybind11 modules. I'm still looking into that.

## Related Issue

Fixes #7113
And more CtrlPorts issues:
- Probes not working (module not found issue)
- Complex value display in gr-ctrlport-monitor causing a crash

## Which blocks/areas does this affect?

The Ctrlpor
closed 2024-02-22 by willcode