gnuradio/gnuradio #7115: Unable to choose editor for Python block on MacOS

techfoamer, 2024-02-06:
### What happened?

Message on console when clicking "Open in Editor" or "Choose Editor" is:
GLib-GIO-WARNING **: 18:00:23.566: No default handler found for content type 'dyn.agq81k3p2su11upputf4gu55s'

Observed this on MacOS 14.3 on both Intel and Apple silicon.

With a mod recommended in https://github.com/ryanvolz/radioconda/issues/86 console now outputs:
">>> Error opening an external editor. Please select a different editor."

### System Information

OS: MacOS14.3
GR Installation Method: radioconda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Place a Python block on a flowgraph.  Open Properties for this block.  Click "Open in Editor".

ryanvolz, 2024-02-06:
I plan on doing a PR with the (partial) fix that I already have, but it would be nice to also roll in whatever additional fixes are needed.

Actually @techfoamer for debugging, it would probably be helpful if you were able to remove [these lines](https://github.com/gnuradio/gnuradio/blob/main/grc/gui/canvas/flowgraph.py#L126-L132) from `grc/gui/canvas/flowgraph.py` so we can see what the actual exception is that it's hitting.

marcusmueller, 2024-02-06:
Might be related to this report on Windows: https://stackoverflow.com/questions/77947451/gnuradio-companion-unable-to-select-an-appropriate-editor-to-edit-python-code

techfoamer, 2024-02-07:
I removed the exception handling around editor.open_editor() at about line 125 in grc/gui/canvas/flowgraph.py.  Here's the output:

Traceback (most recent call last):
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 227, in open_editor
    self.param.parent_flowgraph.install_external_editor(
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 124, in install_external_editor
    editor.open_editor()
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/external_editor.py", line 37, in open_editor
    proc = subprocess.Popen(args=(self.editor, self.filename))
  File "/Users/royce/radioconda/lib/python3.10/subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Users/royce/radioconda/lib/python3.10/subprocess.py", line 1863, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'TextEdit'

ryanvolz, 2024-02-07:
Alright, that looks like it's just getting the name of the editor and not a path to launch it. Can you try making this change?:
```
--- a/grc/gui/Dialogs.py
+++ b/grc/gui/Dialogs.py
@@ -427,6 +427,6 @@ def choose_editor(parent, config):
     response = dialog.run()
     if response == Gtk.ResponseType.OK:
         appinfo = dialog.get_app_info()
-        editor = config.editor = appinfo.get_executable()
+        editor = config.editor = appinfo.get_commandline()
     dialog.destroy()
     return editor
```

(The documentation for `Gio.AppInfo.get_executable()` says that it returns a file system path, but it also confusingly says that it returns a string containing the application's name so this would appear to be the latter...)

techfoamer, 2024-02-07:
...after changes in grc/gui/Dialogs.py

Traceback (most recent call last):
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 227, in open_editor
    self.param.parent_flowgraph.install_external_editor(
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 113, in install_external_editor
    Dialogs.choose_editor(parent, config))  # todo: pass in parent
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/Dialogs.py", line 433, in choose_editor
    editor = config.editor = appinfo.get_commandline()
  File "/Users/royce/radioconda/lib/python3.10/site-packages/gnuradio/grc/gui/Config.py", line 78, in editor
    self._gr_prefs.set_string('grc', 'editor', value)
TypeError: set_string(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.prefs, section: str, option: str, val: str) -> None

Invoked with: <gnuradio.gr.gr_python.prefs object at 0x119c388b0>, 'grc', 'editor', None

ryanvolz, 2024-02-07:
That's fun. Apparently the result of `get_commandline()` is `None`!

I really hate GTK. I'm not finding anything in their documentation that does what we need once we have an AppInfo.

How about this?:
```
--- a/grc/gui/Dialogs.py
+++ b/grc/gui/Dialogs.py
@@ -427,6 +427,7 @@ def choose_editor(parent, config):
     response = dialog.run()
     if response == Gtk.ResponseType.OK:
         appinfo = dialog.get_app_info()
-        editor = config.editor = appinfo.get_executable()
+        exe = find_executable(appinfo.get_executable())
+        editor = config.editor = exe
     dialog.destroy()
     return editor
```

kaylacomer, 2024-03-13:
Output of find_executable was also None - may not work on mac?

Using the full executable path worked for me with one editor (kate), but the other IDEs I have on my computer didn't show up in the list of selectable applications. Notes and TextEdit didn't work. Choosing other non-apple apps in the list (Excel, Chrome) opened the app, so presumably this would work with some other IDEs
```python
    editor = None
    response = dialog.run()
    if response == Gtk.ResponseType.OK:
        appinfo = dialog.get_app_info()
        editor = config.editor = appinfo.get_executable()
        editor = '/Applications/' + editor + '.app/Contents/MacOS/' + editor
        # editor = '/Applications/kate.app/Contents/MacOS/kate'
    dialog.destroy()
    return editor
```

kongmunist, 2024-05-08:
I had a similar error, except none of the editors even showed up in the Pick Editor Dialog. I used @kaylacomer's workaround and directly assigned the `editor` variable to the path of the editor I wanted to use, commenting out everything else. I sense this is not a good workaround but the Python script opens when I hit "Open in Editor" now.
[image]

ArlinJae, 2024-05-17:
Has a change been pushed to the production packages yet? If not, the best course of action here would be to clone the repo and to compile it from scratch, correct?

LsmithOHC, 2024-12-13:
I am also having this issue. 
1. add python module
2. click choose editor
3.  get this error in console: (gnuradio-companion:80033): GLib-GIO-WARNING **: 10:57:36.925: No default handler found for content type 'dyn.agq81k3p2su11upputf4gu55s'.
4. click view all applications same error as above.

I used radioconda to install GNURadio / other libs

Khorishchenko, 2025-01-13:
> I had a similar error, except none of the editors even showed up in the Pick Editor Dialog. I used @kaylacomer's workaround and directly assigned the `editor` variable to the path of the editor I wanted to use, commenting out everything else. I sense this is not a good workaround but the Python script opens when I hit "Open in Editor" now. <img alt="image" width="644" src="https://private-user-images.githubusercontent.com/29759597/328986969-5232870f-c524-4bbb-85bb-25e7eec9aa32.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3NjkxMzEsIm5iZiI6MTczNjc2ODgzMSwicGF0aCI6Ii8yOTc1OTU5Ny8zMjg5ODY5NjktNTIzMjg3MGYtYzUyNC00YmJiLTg1YmItMjVlN2VlYzlhYTMyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDExNDcxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmMTliZDQwNDdjZGIwNzEwMWU3ZmU5Nzc0MDFjZjNmOTdhOGI4MWM4MTcxZDlmOWE5ODVkYzA0ZjdkMTY4OWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.K8uJWQ5mYzowOw96YfMF4f8kBoombcV3uDGKLVYjylc">

This helped me, now when I click to open the editor, Sublime Text opens. Thank you!

ffu, 2025-03-26:
A suitable workaround is to use config.conf to select the editor. Under [grc], set editor = <full path of editor>

First issue is that Apple does not recognize "text/x-python" as a MIME. Apple uses UTI, and if I change to "text/x-python-script", I could get the menu of editors. Second issue is that the AppInfo method get_executable or get_commandline will not return the full executable path of the chosen app. This works for Linux, but not Apple.

So, one possible fix is to check for MacOS and if so, search the /Application directory to find the correct executable path. The other solution is to do it the "right" way and use AppInfo's launch method. This requires some fairly significant code changes, because GRC currently uses a subprocess to run the app.

commit 721e477c (2025-08-07): grc: allow macos editor to be chosen for embedded python blocks

Fixes #7115
closed 2025-08-07 by haakov