gnuradio/gnuradio #7116: Open example does not work

dkozel, 2024-02-06:
### What happened?

Opening an example from the "Examples" tab in a block's properties does not work.

### System Information

OS: Ubuntu
GR Installation Method: source 63bdb24e5 63bdb24e5

### GNU Radio Version

QT-based Companion branch (feature-grc-qt)

### Specific Version

v3.11.0.0git-655-g63bdb24e

### Steps to Reproduce the Problem

Open any block's properties, view the examples list, and select an example.
Double click or press Open Example.

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 133, in <lambda>
    self.open_ex_button.clicked.connect(lambda: self.open_example())
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/dialogs.py", line 172, in open_example
    self._block.parent.app.MainWindow.open_example(ex.text())
AttributeError: 'Flowgraph' object has no attribute 'app'
```

haakov, 2024-02-06:
Ah, regression from last week's refactor. Should be fixed with https://github.com/gnuradio/gnuradio/commit/130fbd2bd3f95144e41e068f0f897759c465a626

dkozel, 2024-02-06:
Fixed! Thanks. Figured it was likely to be a regression.

closed 2024-02-06 by dkozel