gnuradio/gnuradio #7117: Qt GRC Examples browser feedback

dkozel, 2024-02-06:
Feedback on 130fbd2bd

The overall functionality is working well for the examples included with the core. There are a few visual changes that I think would be useful.

- [ ] Unpopulated Title, Author, and Description fields should show as blank rather than "TITLE" etc.
- [ ] Anchor the Description field to the above fields, insert a fixed height (invisible?) divider between the metadata and Description?
- [ ] Add "Language" and "Type" fields showing the language and "Generate Options" values.
- [ ] The image is cool, but was not intuitive to me. Lets get more opinions on it after adding the fields above.
- [ ] Increase width of the flowgraph filenames list and add a way for a user to increasing it (or make it content length adaptive up to some max?
- [ ] Anchor the layout to the left side of the browser window. Probably allow the flowgraph name list and example metadata layouts to increase in width as the window is made wider and narrower?
- [ ] `py_cmd_fg.png`'s dropshadow is too large and has a harsh cutoff on the bottom edge

[image]

dkozel, 2024-02-06:
If every example had a `Title` I'd probably say that the browser should show a list of the flowgraphs by title. As is, it could be a list by title or by flowgraph filename if the title is empty? Thoughts very welcome!

dkozel, 2024-02-07:
[image]

For the Block Properties dialog based Examples tab, it's working great so again the feedback is visual. I think the layout of full filepaths is a bit clunky less useful than the Examples Browser. Would it be a large change to reuse the layout of the Examples Browser inside this tab? I note in there is a function to filter the list already.

https://github.com/gnuradio/gnuradio/blob/130fbd2bd3f95144e41e068f0f897759c465a626/grc/gui_qt/components/example_browser.py#L131-L133

An alternative option could be to display the examples in a TreeView so most of the path is only printed once. I'd say prune the install prefix off, but hopefully this window will handle examples from OOT modules in the future and we'll need to decide how to add and display those.
```
/home/dkozel/prefixes/grmain/share/gnuradio/examples
  qt-gui
    gtgui_message_inputs.grc
    gtgui_multi_input.grc
    ....
  digital
    mpsk_stage6.grc
    ...
```
[image]

dkozel, 2024-02-07:
Could the Examples Browser be launched with a filter value from the right click context menu of a block?

haakov, 2024-02-07:
>     * [ ]  The image is cool, but was not intuitive to me. Lets get more opinions on it after adding the fields above.

I'm not a huge fan of it either. I just thought the window looked a bit empty without it. I wouldn't mind replacing it though.

>     * [x]  Increase width of the flowgraph filenames list and add a way for a user to increasing it (or make it content length adaptive up to some max?
>     * [x]  Anchor the layout to the left side of the browser window. Probably allow the flowgraph name list and example metadata layouts to increase in width as the window is made wider and narrower?

Qt's widget sizepolicy system is a bit intricate, but I've tweaked the widths and made the borders adjustable

haakov, 2024-02-07:
> If every example had a `Title` I'd probably say that the browser should show a list of the flowgraphs by title. As is, it could be a list by title or by flowgraph filename if the title is empty? Thoughts very welcome!

I agree, titles are better than file names. I've now updated it so that the title is used if there is one (filename otherwise).

haakov, 2024-02-07:
> For the Block Properties dialog based Examples tab, it's working great so again the feedback is visual. I think the layout of full filepaths is a bit clunky less useful than the Examples Browser. [...]

That's very doable, and not a large change! 

But that would suddenly make the block properties dialog very wide. I think a tree view might be better suited for this kind of display (even though that would still make the dialog a bit wider)

haakov, 2024-02-10:
Switched to a tree view:
[image]

The example browser can now be opened from the right click menu of a block, the block tree, and the File menu

dkozel, 2024-02-10:
Opening an OOT supplied example results in a blank canvas (no blocks).

```
Traceback (most recent call last):
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/window.py", line 947, in open_example
    self.open_triggered(example_path, False)
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/window.py", line 938, in open_triggered
    self.currentFlowgraphScene.import_data(initial_state)
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/canvas/flowgraph.py", line 354, in import_data
    self.core.import_data(data)
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/core/FlowGraph.py", line 476, in import_data
    self.new_block(block_id='_dummy',
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/core/FlowGraph.py", line 344, in new_block
    block = self.parent_platform.make_block(self, block_id, **kwargs)
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/core/platform.py", line 451, in make_block
    return cls(parent, **kwargs)
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/canvas/block.py", line 34, in __init__
    super(self.__class__, self).__init__(parent)
TypeError: DummyBlock.__init__() missing 2 required positional arguments: 'missing_block_id' and 'parameters'
```

haakov, 2024-02-10:
> Opening an OOT supplied example results in a blank canvas (no blocks).
>
> ```
> Traceback (most recent call last):
>   File [...]

Should be fixed now