gnuradio/gnuradio #7118: Qt Preferences menu feedback

dkozel, 2024-02-07:
A start on feedback for the Preferences Menu. I did a quick scan for what preferences are in the main codebase using `ag 'prefs::singleton' -A2` and at least the ones in `gnuradio-runtime` seem reasonable to include in these pages. We have default .conf files installed in `etc/gnuradio/conf.d`. How are those generated? Is there a mechanism we can latch into to identify all possible settings and their values?

GRC preferences

- [ ] Add option to start GRC Qt or GTK by default. (Allow override by `--qt` or `--gtk` flags)
- [ ] Default Text editor for the Python Block/Module blocks
- [ ] Log Level (does GRC have it's own log levels?)

Appearance

- [ ] Display Wiki panel?
- [ ] Remove main GRC menu `Tools > Set Default Qt GUI Theme` option as redundant

GNU Radio preferences

- [ ] Log Level
- [ ] Log file location
- [ ] `debug_level` seems to be different than the log level. File vs console?
- [ ] ControlPort and PerfCounters
- [ ] `max_messages`

There are a number of other settings in our core blocks, most notably a variety of audio settings. I don't have a good suggestion on how we should deal with these. I believe that a block should be able to register a `setting` or `preference` with name, type, default, and description data just like we currently have `params`. This feels like an extension to the `block.yml` schema? Some settings/prefs may be separate from any particular block though. It may (likely) be out of scope to solve this for 3.10, but should be an architecture discussion for 4.0 at minimum.

dkozel, 2024-02-07:
From @willcode in chat
> A block path editor, allowing multiple paths, could be useful. Haven't thought much about that one though. Hier path, additional block paths, ...

I'd add configuration file/directory path to that.

haakov, 2024-02-08:
Output from `ag 'prefs::singleton' -A2`:
https://gist.github.com/haakov/fc22222caecd78e7f44cf2c940a4a231

dkozel, 2024-02-10:
- [ ] The preferences list layout adjusts vertical padding between items based on how tall the GRC Preferences dialog is. Currently that means it can be over-compressed causing text to look bad. Forcing a fixed layout is probably fine and prevents issues. The current maximum looks good to me on my screen. Looks like you were already looking at this.
https://github.com/gnuradio/gnuradio/blob/48070b4d0d127dece49347ea8778deb48eb0beab/grc/gui_qt/components/preferences.py#L77-L79

[image]
[image]

- [ ] Preferences dialog should have `Save` instead of `Ok`. Disable `Save`

haakov, 2024-02-12:
> From @willcode in chat
> 
> > A block path editor, allowing multiple paths, could be useful. Haven't thought much about that one though. Hier path, additional block paths, ...
> 
> I'd add configuration file/directory path to that.

I'm not 100 % sure how that should work in practice. I suppose it's kind of a chicken and egg situation if the configuration file specifies where to find the configuration file?

haakov, 2024-02-13:
> GRC preferences
> 
>     * [ ]  Add option to start GRC Qt or GTK by default. (Allow override by `--qt` or `--gtk` flags)
> 
>     * [ ]  Default Text editor for the Python Block/Module blocks
> 
>     * [ ]  Log Level (does GRC have it's own log levels?)

Note: I've tried to get GRC Gtk and GRC Qt to use the same config file. (I'm not sure if this is a good idea to begin with, but I thought it'd be neat if it worked seamlessly) 

It seems to work at first, but they use different formats and at some point GRC Gtk will re-save all of GRC Qt's values as lists, which are then parsed as strings by GRC Qt. Chaos ensues.

I've decided to keep the GRC Qt settings in grc_qt.conf, and make the `gnuradio-companion` script check if `default_grc` is set to `grc_qt` or `grc_gtk` in `grc_qt.conf`. Commit: https://github.com/gnuradio/gnuradio/commit/6a410ae8cc29ff24b98e1c6e8334b69ce67cc6d2