gnuradio/gnuradio #7120: spdlog/fmt error for 3.10.9.2

keukenrol, 2024-02-08:
### What happened?

I required UHD version 4.6 with GU Radio, thus I was following the guide to build from source linked in system infomation. As soon as I run the command "make -j3" I get the following errors:

`gnuradio/build/gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx:5,
                 from <command-line>:
/usr/include/spdlog/common.h:127:111: error: ‘basic_runtime’ is not a member of ‘fmt’
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                               ^~~~~~~~~~~~~
/usr/include/spdlog/common.h:127:125: error: template argument 2 is invalid
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>
      |                                                                                                                             ^~~~
/usr/include/spdlog/common.h:127:138: error: expected ‘{’ before ‘>’ token
  127 |           std::is_convertible<T, fmt::basic_string_view<Char>>::value || std::is_same<remove_cvref_t<T>, fmt::basic_runtime<Char>>::value>`
Has this anything to do with the package versions?

### System Information

OS: (Ubuntu 22.04.3 LTS)
GR Installation Method: (LinuxInstall guide: https://wiki.gnuradio.org/index.php?title=LinuxInstall#Installing_Volk)
gcc 11.4.0-1ubuntu1~22.04
libfmt-dev 8.32
libspdlog 1:1.9.2+ds-0.2
UHD 4.6.0.0

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Follow the linuxinstall guide on the linked page. When executing the make command, it will throw the errors.

willcode, 2024-02-08:
From the error, this looks like a version mismatch between spdlog and libfmt. There have been similar things that @marcusmueller was able to fix within GR. Not sure about this one.

keukenrol, 2024-02-12:
Did a reboot and then a sudo make which fixed it.

closed 2024-02-12 by keukenrol

marcusmueller, 2024-02-12:
@keukenrol hope you really mean `make clean && make && sudo make install`, because you should never build as root.