gnuradio/gnuradio #7121: RFNoC Radio usage

edombek, 2024-02-09:
### What happened?

A bug happened!

### System Information

OS: (Archlinux)
GR Installation Method: pacman package

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

created the simplest scheme using RFNoC
[image]
Generate python script or run

### Relevant log output

```shell
Generating: "/tmp/untitled.py"
Generate Error: Undefined
>>> Failure
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/gnuradio/grc/gui/Application.py", line 758, in _handle_action
    generator.write()
  File "/usr/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 86, in write
    for filename, data in self._build_python_code_from_template():
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 135, in _build_python_code_from_template
    connections=self._connections(),
                ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/gnuradio/grc/core/generator/top_block.py", line 328, in _connections
    code = template.render(
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mako/template.py", line 438, in render
    return runtime._render(self, self.callable_, args, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/mako/runtime.py", line 874, in _render
    _render_context(
  File "/usr/lib/python3.11/site-packages/mako/runtime.py", line 916, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3.11/site-packages/mako/runtime.py", line 943, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x7855997c2350", line 32, in render_body
  File "/usr/lib/python3.11/site-packages/mako/runtime.py", line 230, in __str__
    raise NameError("Undefined")
NameError: Undefined
```

edombek, 2024-02-09:
This problem occurs when using the "rx streamer" block. Perhaps I did not understand the output logic from the device a little

basti-schr, 2024-04-19:
Can confirm also with the "Tx Streamer" Block.

System Information

OS: Ubuntu 22.04.4
GR Installation Method: self compiled
GNU Radio Version: 3.10.9.2 (Phython 3.10.12)

Flowgraph: Null Source -> RFNoC Tx Streamer -> RFNoC TX Radio

```bash
Generate Error: Undefined
>>> Failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/gui/Application.py", line 758, in _handle_action
    generator.write()
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/generator/top_block.py", line 86, in write
    for filename, data in self._build_python_code_from_template():
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/generator/top_block.py", line 135, in _build_python_code_from_template
    connections=self._connections(),
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/grc/core/generator/top_block.py", line 328, in _connections
    code = template.render(
  File "/usr/lib/python3/dist-packages/mako/template.py", line 476, in render
    return runtime._render(self, self.callable_, args, data)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 878, in _render
    _render_context(
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 920, in _render_context
    _exec_template(inherit, lclcontext, args=args, kwargs=kwargs)
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 947, in _exec_template
    callable_(context, *args, **kwargs)
  File "memory:0x77e90a02efb0", line 32, in render_body
  File "/usr/lib/python3/dist-packages/mako/runtime.py", line 230, in __str__
    raise NameError("Undefined")
NameError: Undefined
```

It would be really helpful to use RFNoC from GR.

mbr0wn, 2025-01-28:
This is a dupe of #7313.

closed 2025-01-28 by mbr0wn