gnuradio/gnuradio #7123: OOT Module Browser feedback

dkozel, 2024-02-10:
- [ ] Needs increased error handling for missing fields (can we setup a formal schema to check against?)
```
Traceback (most recent call last):
  File "/home/dkozel/prefixes/grmain/lib/python3.10/site-packages/gnuradio/grc/gui_qt/components/oot_browser.py", line 65, in populate_right_view
    self.license_label.setText(f"<b>License:</b> {module['license']}")
KeyError: 'license'
```

- [ ] Formatting error of GNU Radio version list (gr-satellites shown here)
> Supported GNU Radio Versions: v; 3; .; 8; ,; ; v; 3; .; 9; ,; ; v; 3; .; 1; 0

- [ ] Version shows as "NONE" when manifest is missing it. Add a check to leave that blank. Separately/additionally I don't think we should rely on the Manifest for the version number. People won't update it, something should be written by CMake using the authoritative version number from CMake. Thoughts/ideas? Is this already available in code?
https://github.com/gnuradio/gnuradio/blob/69ae692d14cd12cf23cb0e1ea20f74128b12e08e/grc/gui_qt/components/oot_browser.py#L51C67-L51C74

haakov, 2024-02-14:
>     * [ ]  Version shows as "NONE" when manifest is missing it. Add a check to leave that blank. Separately/additionally I don't think we should rely on the Manifest for the version number. People won't update it, something should be written by CMake using the authoritative version number from CMake. Thoughts/ideas? Is this already available in code?
>       https://github.com/gnuradio/gnuradio/blob/69ae692d14cd12cf23cb0e1ea20f74128b12e08e/grc/gui_qt/components/oot_browser.py#L51C67-L51C74
> 
>     * [ ]  MANIFEST.md needs to be installed by CMake

PR: https://github.com/gnuradio/gnuradio/pull/7134

This is an idea I had, I'm not sure it's a good one yet :) 

The method used in the PR means that each OOT module gets its own directory in `$PREFIX/share/gnuradio/manifests`. The manifest gets installed with the version as part of the file name, so that it's easy to see which version we're using. This way, we also avoid overwriting old versions of the manifest with new versions.

haakov, 2024-02-14:
> * [ ]  Needs increased error handling for missing fields (can we setup a formal schema to check against?)

I've added a [validation routine](https://github.com/gnuradio/gnuradio/blob/ff760449dffea08befab9b5363730aa85ca5e766/grc/gui_qt/components/oot_browser.py#L49), although I'm not sure what the rules should be. Maybe that's part of a larger discussion?