gnuradio/gnuradio #7126: qa_modtool doesn't skip if pygccxml is not available

marcusmueller, 2024-02-12:
### What happened?

Currently, the win and Mac conda CI runners consistently fail, because qa_modtool fails.

### System Information

OS: CI conda mac, win
GR Installation Method: CI

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

make pr

### Relevant log output

```shell
243/265 Test #242: test_modtool .................................***Failed    0.43 sec
Traceback (most recent call last):
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/core/parseheader_generic.py", line 26, in <module>
    from pygccxml import parser, declarations, utils
ModuleNotFoundError: No module named 'pygccxml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "$SRC_DIR/gr-utils/modtool/tests/test_modtool.py", line 25, in <module>
    from gnuradio.modtool.core import ModToolNewModule
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/core/__init__.py", line 12, in <module>
    from .add import ModToolAdd
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/core/add.py", line 20, in <module>
    from gnuradio.bindtool import BindingGenerator
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/bindtool/__init__.py", line 1, in <module>
    from .core.generator import BindingGenerator
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/bindtool/core/generator.py", line 11, in <module>
    from gnuradio.blocktool import BlockHeaderParser, GenericHeaderParser
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/__init__.py", line 11, in <module>
    from .core.parseheader_generic import GenericHeaderParser
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/core/parseheader_generic.py", line 30, in <module>
    from ...modtool.cli import ModToolException
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/cli/__init__.py", line 11, in <module>
    from .base import cli, common_params, block_name, ModToolException
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/cli/base.py", line 17, in <module>
    from pkg_resources import iter_entry_points
ModuleNotFoundError: No module named 'pkg_resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/core/parseheader_generic.py", line 26, in <module>
    from pygccxml import parser, declarations, utils
ModuleNotFoundError: No module named 'pygccxml'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "$SRC_DIR/gr-utils/modtool/tests/test_modtool.py", line 33, in <module>
    from modtool.core import ModToolNewModule
  File "$SRC_DIR/gr-utils/modtool/core/__init__.py", line 12, in <module>
    from .add import ModToolAdd
  File "$SRC_DIR/gr-utils/modtool/core/add.py", line 20, in <module>
    from gnuradio.bindtool import BindingGenerator
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/bindtool/__init__.py", line 1, in <module>
    from .core.generator import BindingGenerator
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/bindtool/core/generator.py", line 11, in <module>
    from gnuradio.blocktool import BlockHeaderParser, GenericHeaderParser
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/__init__.py", line 11, in <module>
    from .core.parseheader_generic import GenericHeaderParser
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/blocktool/core/parseheader_generic.py", line 30, in <module>
    from ...modtool.cli import ModToolException
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/cli/__init__.py", line 11, in <module>
    from .base import cli, common_params, block_name, ModToolException
  File "$PREFIX/lib/python3.11/site-packages/gnuradio/modtool/cli/base.py", line 17, in <module>
    from pkg_resources import iter_entry_points
ModuleNotFoundError: No module named 'pkg_resources'
```

marcusmueller, 2024-02-12:
@mormj  @willcode @ryanvolz  is `pkg_resources` a dependency we're not explicitly stating? I was under the impression it's part of `python3-setuptools`, but doesn't seem to be there on the win and mac os conda CI!

willcode, 2024-02-12:
Since it's deprecated

https://setuptools.pypa.io/en/latest/pkg_resources.html

Some distros might have actually dropped it.

ryanvolz, 2024-02-12:
It *is* part of `setuptools`, but currently the conda recipe doesn't depend on `setuptools` during the build and cmake test phase. So that explains why its failing. It's currently a runtime dependency to allow modtool to work at runtime: https://github.com/gnuradio/gnuradio/blob/main/.conda/recipe/meta.yaml#L120-L121. I guess something used to pull in setuptools incidentally and now it doesn't? Probably because packages are dropping it due to deprecation as @willcode notes.

It can be added as a dependency in the `host` environment explicitly or the code can be changed.

marcusmueller, 2024-02-12:
hm, don't have the time right now, but I guess the rightest approach would be to replace the need for pkg_resources with whatever supersedes it for our use case. But I can already see the bug reports coming up where a slight change in behaviour breaks differently on 5 platforms… sigh.

@ryanvolz for the meantime, I think if adding that would be possible, that'd be great, as currently our CI never passes because of this.

ryanvolz, 2024-02-12:
Alright, see #7128.

PR #7128 (2024-02-13): ci: conda: Fix missing pkg_resources during modtool test

## Description

Add a setuptools dependency in the conda recipe so that tests requiring `pkg_resources` don't fail.

## Related Issue

Fixes #7126.

## Which blocks/areas does this affect?

## Testing Done
closed 2024-02-13 by willcode