gnuradio/gnuradio #7132: QT GUI Constellation in radioconda need sample per symbol

mohammadsdtmnd, 2024-02-14:
### What happened?

This block needs a parameter named: sample per symbol. I wonder how this big bug could be ... !

### System Information

OS: Windows
GR Installation Method: RadioConda

### GNU Radio Version

3.11-git (main)

### Specific Version

3.11.7.0

### Steps to Reproduce the Problem

Open this block, and you wont see that parameter.
If you do GMSK with sample per symol higher that 1, the constellation output will be wrong.

willcode, 2024-02-14:
GMSK blocks have `Samples/Symbol` and it defaults to `2`.

mohammadsdtmnd, 2024-02-14:
> GMSK blocks have `Samples/Symbol` and it defaults to `2`.
Updated, sorry for wrongly questioning!
I meant QT GUI Constellation doesn't have this parameter

willcode, 2024-02-14:
This sink is just a x/y plot. It wouldn't know what to do with samples/symbol. It is common to see effects of oversampling on a constellation plot, too. If you really want to see one sample/symbol, try putting a decimating filter before the sink.

If you have a good description of a feature you'd like to see, please change the title/description and reopen.

closed 2024-02-14 by willcode

mohammadsdtmnd, 2024-02-23:
> This sink is just a x/y plot. It wouldn't know what to do with samples/symbol. It is common to see effects of oversampling on a constellation plot, too. If you really want to see one sample/symbol, [...]

The constellation interpret the data to symbols and view them on the plot, doesn't it? it must be aware that each symbol takes how much sample?

willcode, 2024-02-23:
No, a constellation plot does not convert to symbols. It just plots I/Q data in X/Y form.

mohammadsdtmnd, 2024-03-01:
> No, a constellation plot does not convert to symbols. It just plots I/Q data in X/Y form.

Then why simulink constellation viewer have this?

willcode, 2024-03-01:
The simulink constellation diagram includes a downsampler, but it does not try to interpret symbols/samples. It does plot a reference constellation, which is a nice idea. It also has a "Trajectory" setting which probably results in the same display you see in GNU Radio.

The GR Constellation Sink does not include a downsampler. It would be added in front of the sink if desired.

Some simulink features might be nice to have in GR if someone want to work on them.