gnuradio/gnuradio #7133: Race condition and deadlock in TCP Sink server mode

argilo, 2024-02-14:
While investigating #6699, @marcusmueller noticed that the following code often hangs:

```python
from gnuradio import network

t_sink = network.tcp_sink(1, 1, "localhost", 1234, 2)
t_sink.start()
t_sink.stop()
```

GDB shows that `stop()` gets stuck here waiting for the listener thread to terminate:

https://github.com/gnuradio/gnuradio/blob/7d61746e27778c56eb8a805ad940c89ff265e313/gr-network/lib/tcp_sink_impl.cc#L206

Meanwhile, the listener thread continues executing `run_listener()` indefinitely, because `d_stop_thread` is not set.

This happens because `start()` kicks off `run_listener()` in a background thread:

https://github.com/gnuradio/gnuradio/blob/7d61746e27778c56eb8a805ad940c89ff265e313/gr-network/lib/tcp_sink_impl.cc#L98-L102

Some time later, `run_listener()` begins executing and sets `d_thread_running` to `true`:

https://github.com/gnuradio/gnuradio/blob/7d61746e27778c56eb8a805ad940c89ff265e313/gr-network/lib/tcp_sink_impl.cc#L108-L123

But `stop()` may be called before the thread begins executing, and it skips setting `d_stop_thread` if `d_thread_running` is not yet `true`:

https://github.com/gnuradio/gnuradio/blob/7d61746e27778c56eb8a805ad940c89ff265e313/gr-network/lib/tcp_sink_impl.cc#L185-L189

As a result, the listener thread never exits, leading to deadlock.

One possible fix would be to set `d_thread_running` to `true` in `start()` instead of deferring it to the background thread.

marcusmueller, 2024-02-14:
How possible is it that we're using this async API incorrectly if we have a separate listener thread?

argilo, 2024-02-14:
That sounds plausible. I don't know enough about Asio to know for sure.

marcusmueller, 2024-02-14:
@mboersch would you care to have a look at `tcp_sink_impl`?

willcode, 2024-02-15:
The A in ASIO sort of hints that there shouldn't be a listener thread.