gnuradio/gnuradio #7153: Mismatch in control_loop damping factor and comment

dkozel, 2024-02-27:
### What happened?

@AlexandreRouma reports that the damping factor of `sqrtf(2.0f) / 2.0f` results in an underdamped loop rather than critically damped as the comment indicates. They found that the damping factor should be `1` to be critically damped.

https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/lib/control_loop.cc#L27-L28

This control_loop code is used in many PLLs around GNU Radio so changing the factor would affect many areas, possibly positively possibly negatively.

If someone can confirm that the comment and current damping factor disagree we should at least update the comment ASAP with a note and fixme.

### System Information

OS: All
GR Installation Method: source

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

@AlexandreRouma has mathematical analysis and simulations.

willcode, 2024-02-27:
@daniestevez write https://destevez.net/2017/09/about-critical-damping/ so would probably know!

daniestevez, 2024-02-27:
I'll try to read this in more detail tomorrow, but my immediate take is that different people meant different things by critical damping (I think I've seen it used to refer to "both roots are real and equal" and also to "both roots are of the form $-a \pm ai$"), and that in any case these notions and formulas come from continuous-time loops and are only an approximation to the behaviour of discrete-time loops (increasing worse as `BT` grows). [This paper](https://www.semanticscholar.org/paper/Controlled-root-formulation-for-digital-loops-Stephens-Thomas/1511c6aa1110b841f646cbc6f5c385e7d861268c) is quite good because it does all the analysis directly in discrete time and also relates the results to the continuous-time approximation (unfortunately the exact formulas that come out of the discrete case aren't that simple).

daniestevez, 2024-02-28:
Looking at Tom Rondeau's post I see that the formulas for $\alpha$ and $\beta$ were not chosen in terms of the loop noise bandwidth ($B_L$ in the paper I'm using), but rather in terms of the undamped natural frequency (more specifically as one half of this; he has the formula $\theta_n = \varpi_n T_s/2$).

In the notation I used above, the undamped natural frequency is $\sqrt{\kappa_2}$ (it is the imaginary part of the loop roots if we force $\kappa_1 = 0$, thus removing damping). In the continuous time approximation, $\sqrt{\kappa_2} = 2 B$, so indeed, in these coefficient design formulas, $B$ is set to equal one half of the undamped natural frequency, rather than the loop bandwidth.

daniestevez, 2024-02-28:
Another thing we can look at is what happens to the roots in the discrete-time formulation. Using equations (45) and (49) from the paper, the loop roots $z_1$ and $z_2$ satisfy
```math
z_1 + z_2 = 2 - K_1 - K_2
```
```math
z_1 z_2 = 1 - K_1
```
Since the roots are the roots of the equation $z^2 - (z_1 + z_2)z + z_1z_2$, we have
```math
z = \frac{-K_1 - K_2 + 2 \pm \sqrt{(K1 + K_2 - 2)^2 - 4(1 - K_1)}}{2}.
```
If my pen and paper calculations are right, the term inside the square root is equal to
```math
\frac{16B^2T^2}{\Delta^2}(\zeta^2-1),
```
where
```math
1 + 2\zeta B T + B^2T^2.
```

This means that the two roots are equal if and only if $\zeta = 1$, so again we recover the same condition for the critically damped case.

The analysis for the standard underdamped case is much more complicated. The roots are now in the $z$-plane rather than the $s$-plane, and need to be translated to concepts in the $s$-plane by means of an exponential map. More precisely, equation (50) in the paper gives the roots as $z = e^{-\beta_1(1\pm\eta_1)T}$. For the standard underdamped case, $\eta_1 = i$.

Again assuming no mistakes in my pen and paper calculations, the roots can be simplified to
```math
z = \frac{1 - B^2T^2 \pm 2B^2T^2 \sqrt{\zeta^2 - 1}}{\Delta}.
```
Under the assumption that $0 < \zeta < 1$, the modulus squared of $z$ is
```math
|z|^2 = \frac{(1-B^2T^2)^2 + 4B^2T^2(1-\zeta^2)}{\Delta^2}.
```
This must equal $e^{-2\beta_1T}$, so
```math
\beta_1 T = -\frac{1}{2}\log\frac{(1-B^2T^2)^2 + 4B^2T^2(1-\zeta^2)}{\Delta^2}.
```
Since $\eta_1 = i$, it follows that $\text{Im}(z)/\text{Re}(z)$ must equal $\pm \tan(\beta_1T)$. So we have
```math
\pm \tan(\beta_1 T) = \frac{\text{Im}(z)}{\text{Re}(z)} = \frac{\pm 2B^2T^2\sqrt{\zeta^2 - 1}}{1 - B^2T^2}
```
Replacing the above formula for $\beta_1 T$ here, we get a fairly complicated equation that can be solved (numerically, I guess) for $\zeta$.

AlexandreRouma, 2024-02-28:
I guess I should have let derek post my handwritten analysis and simulations sooner so this would have been more clear so here it is (sorry I write like a doctor, this was just me messing around a few days ago, didn't plan to post it):

[image]

Also, yes I know I could have just used the canonical expression of a second order LTI system to get the damping factor and natural frequency (which is what they really meant by bandwidth instead of the 3dB bandwidth we usually use). But as I said, I was just messing around and didn't get time this week to rewrite it properly.

Indeed, this is just an issue of definition of what "critically damped" means. In mechanical engineering (my field), we always use this to mean a damping factor of one, and when calling the function, I expected to get a critically damped loop and I didn't. Here is the quick simulation I did with gnuradio's PLL (it's just a standard PLL, nothing out of the ordinary). The blue trace is with a damping factor of 1 and is what I wanted to get when I expected a critically damped loop, while the orange one is what the function returns:

[image]

This is also why I said it wasn't necessarily a bad thing, it just depends what you need. As can be seen in the graph, the sqrt(2)/2 damping factor results in faster locking, so it is usually to be preferred.

Finally something I didn't touch on in my messages with derek is the problem of the error detector gain. For example, the costas loop's error (on top of depending on the amplitude of the signal) has a gain of sqrt(2). Below is my mathematical analysis showing that:

[image]

Why is this important? This means we must divide the gains by the gain of the TED if we want the damping factor and bandwidth to be as expected. I'll note that this is also a problem if anyone attempts to use that function with MM clock recovery, but the gains of that TED are even  more wild and I'm not yet comfortable sharing my math on it because it was a mess.

daniestevez, 2024-02-28:
> For example, the costas loop's error (on top of depending on the amplitude of the signal) has a gain of sqrt(2).

I understand you mean the discriminator for order 4 https://github.com/gnuradio/gnuradio/blob/main/gr-digital/lib/costas_loop_cc_impl.h#L64, since the discriminator for order 2 I think it's fine.

I agree with you. Assuming without loss of generality that the input $I + iQ$ sits in the first quadrant and that it is of the form $$\cos(\pi/4+\varphi) + i \sin(\pi/4 + \varphi)$$ with a small angle error $\varphi$ (since the gain of these formulas can only be expected to be 1 for a small error), then the discriminant gives
```math
Q - I = \sin(\pi/4 + \varphi) - \cos(\pi/4 + \varphi) = \sqrt{2} \sin(\varphi) \approx \sqrt{2} \varphi.
```
I wonder what was the rationale behind this formula, because it does work for a signal that has amplitude $\sqrt{2}/2$ rather than amplitude $1$. But I think that's not the intended use case. I could understand something that works for signals of amplitude $\sqrt{2}$ because someone wants the constellation points to be $\pm 1 + \pm i$ rather than $\pm \sqrt{2}/2 \pm i \sqrt{2}/2$, but I don't see a reason for using a signal of amplitude $\sqrt{2}/2$.

As a said before, users might have flowgraphs that depend on this brokeness by having adjusted the loop bandwidth manually to compensate for this effect (even though this messes up the damping factor). However, in this case I'm in favor of correcting the error, since the formula for the correct discriminant is straightforward and it doesn't depend on approximations or philosophical points of view as it happens with the loop coefficients.

It would also be good to review the order 8 Costas loop discriminator. I did this exercise some time ago, but I didn't remember if the discriminant was correct or if it was biased by a multiplicative constant, as it happens for order 4.

daniestevez, 2024-02-28:
In any case, circling back to @dkozel's original request:

> If someone can confirm that the comment and current damping factor disagree we should at least update the comment ASAP with a note and fixme.

This was more about what to do with the comment than with the current code.

Coming from mathematics I was taught (half seriously, half in jest) that you can use any names you want for things as long as you define precisely what you mean by those names. So I won't argue if it's correct or not to call $\zeta = \sqrt{2}/2$ a critical damping.

What I will say is that often critical damping means $\zeta = 1$ (and this case is of algebraic significance because it is a singular case where you get double roots), but that Tom Roundeau (who I think wrote this code) [mentions](http://www.trondeau.com/blog/2011/8/13/control-loop-gain-values.html)

> In control systems, the damping factor specifies the oscillation of the loop, whether it's under, over, or critically damped. In a normalized system, a damping factor of 0.707 (or the sqrt(2)/2) is a critically damped system. This is a good approximation for most systems, so we can just set it and forget about it.

And Tom is not the only person I've seen calling $\zeta = \sqrt{2}/2$ a critical damping in a DSP context (and $\zeta = \sqrt{2}/2$ doesn't have any special algebraic significance).

So I don't know what to do about the comment. Maybe add a warning note to the comment saying that different people mean different things by critical damping, and that here we mean $\sqrt{2}/2$, which is commonly used in DSP and gives faster convergence than $1$.

daniestevez, 2024-02-28:
Oh, I remember the thing about the overshoot happening even with critical damping in the continuous-update loop.

Put for example $\kappa_1 = 2$, $\kappa_2 = 1$ for simplicity. This gives two roots at $s = -1$. The closed loop transfer function that gives $\hat{\Phi}(s) = H(s)\Phi(s)$ is (see the paper or calculate it)
```math
H(s) = \frac{\kappa_1 s + \kappa_2}{s^2 + \kappa_1 s + \kappa_2}.
```
The loop error is calculated in the s-domain as $\Phi(s) - \hat{\Phi}(s) = (1 - H(s))\Phi(s)$, and we have
```math
1 - H(s) = \frac{s^2}{s^2 + \kappa_1 s + \kappa_2}.
```

Now assume that the input $\varphi(t)$ is equal to the unit step $u(t)$. Its Laplace transform is $\Phi(s) = 1/s$. Therefore, the Laplace transform of the error is
```math
\frac{1 - H(s)}{s} = \frac{s}{s^2 + \kappa_1 s + \kappa_2} = \frac{s}{s^2 + 2s + 1} = \frac{1}{s + 1} - \frac{1}{(s+1)^2}.
```
Taking the inverse Laplace transform, we get that the time domain error is
```math
e^{-t} - te^{-t} = (1-t) e^{-t}.
```
Note that the error changes sign at $t = 1$. This means that there is overshoot.

AlexandreRouma, 2024-02-28:
> I wonder what was the rationale behind this formula

Simple: It's fast. You're just supposed to adjust the loop gains so that the behavior of the loop is as expected

> This was more about what to do with the comment than with the current code.

Exactly, this was my point but it wasn't reflected well through the initial post. Clearly the generates coefficients are fine for their use case but the output of the function was not what I was expecting.

> So I don't know what to do about the comment.

In my opinion, the text "critically damped" should be prefaced by the definition used. 

In any case, there is still the issue of different error detectors having different gains. IMO, to avoid any issue in the code, the function that generates the gains should take as an input the detector gain and correct the gains accordingly (aka, just ki/kd, kp/kd with ki the integral gain, kp the proportional gain and kd the detector gain). Each block should then pass the gain of their detectors when creating critically damped gains.

daniestevez, 2024-02-28:
> Simple: It's fast.

I'm not sure that the current
```c++
return ((sample.real() > 0.0f ? 1.0f : -1.0f) * sample.imag() -
              (sample.imag() > 0.0f ? 1.0f : -1.0f) * sample.real());
```
is any faster than
```c++
const float scale = sqrtf(2.0f) / 2.0f;
return ((sample.real() > 0.0f ? scale : -scale) * sample.imag() -
              (sample.imag() > 0.0f ? scale : -scale) * sample.real());
```

AlexandreRouma, 2024-02-28:
Yes modern compilers are often able to use some smarter math when there are unit products, often turning the entire thing branchless. Check the assembly code for both versions on `-O3`.

marcusmueller, 2024-02-28:
The scaled version *can* be faster than the unscaled version: 

[image]

https://quick-bench.com/q/f4YDk4zVY6mjVxnIrozR2KYRlss

If you look at the disassembly, you'll notice the `copysign` method has *waaaaay* more instructions, but avoids the control flow uncertainty due to the `comiss` (`com`pare a `s`ingle precision with a `s`ingle precision float)/`ja` (`j`ump if strictly gre`a`ter):

https://gcc.godbolt.org/z/5nW3jbEG7

daniestevez, 2024-02-28:
@marcusmueller unless I'm missing something, your implementation with `copysign` doesn't do the same thing. It's not the same to "toggle the sign of Q if I is negative" (the original thing) than it is to "make the sign of Q equal the sign of I" (the copysign thing).

In any case I'm confident that there should be a way to write this discriminant in a branchless manner, and that it will be faster that branching implementations. And I'm also certain that the fastest way to do unscaled will be faster than the fastest way to do scaled, because you simply have one operation less to do.

Maybe another point of view at this is that GNU Radio doesn't try to have the utmost fastest performance if it comes at the cost of other factors (usability, etc.). While I agree that pushing scale factors to the loop coefficients is a good idea for a specialized receiver (rather than something built out of general use blocks), in GNU Radio several general use loop blocks (Symbol Sync, Costas) have an error output that can be used for debugging or benchmarking. This favours the use of properly scaled discriminants, even if the performance is slightly worse.

AlexandreRouma, 2024-02-28:
> In any case I'm confident that there should be a way to write this discriminant in a branchless manner

There is, you simply replace the mantissa and exponent to be 1. This however is very likely what the compiler already optimizes the code to...

daniestevez, 2024-02-28:
> This however is very likely what the compiler already optimizes the code to...

As you can see in the godbolt that Marcus linked, the compiler (gcc with -O3) is doing `comiss` and `ja`, not a branchless implementation.

daniestevez, 2024-02-28:
Now I'm wondering who else besides Tom Rondeau I have seen calling critical damping to $\zeta = \sqrt{2}/2$. I would swear I read it in some book, but can't find it / maybe I'm imagining it.

I Google search for "critical damping loop 0.7" or "critical damping PLL 0.7" yielded these two hits:

https://owolff.com/application-support/active-noise-cancellation/

> It can be calculated using equation 1, that a damping ratio around 0.7 will result in a Q factor of 0.7 (called critical damping).

https://www.edaboard.com/threads/frequency-drift-in-pll-design.132400/

> f). The desired Phase Margin. A good design should have a phase margin of not less than 40 degrees. Note that phase margin is also related to damping factor. For ‘critical damping’, d = 0.7, and this is equivalent to a phase margin of 37 degrees.

I reckon these are a minority compared to all the sources that say that critical damping is $\zeta = 1$. However, this is not the easiest thing to search for in Google. I'm curious to see what the usual text references in communications theory say.

marcusmueller, 2024-02-28:
Ah true! I sadly did a 2×2 sign table in my head and got it totally wrong :) Just because I had already run the benchmarks: If you take my wrong implementation, and shift the scalar multiplication after the difference calculation, it ends up being free:

[image]

https://quick-bench.com/q/VrHjnJifRBXrusrtPzLLONluPGo

daniestevez, 2024-02-28:
Nice find. CPUs are very surprising.

bistromath, 2024-02-28:
Critical damping in a control loop is of course $`\zeta=1`$. I wonder if this DSP-specific definition of "critical damping" where $`\zeta=\sqrt 2/2`$ comes from IIR filter design, in which $`\zeta=\sqrt 2/2`$ gives rise to a filter which has no overshoot -- in the _frequency_ domain!