gnuradio/gnuradio #7156: Build problem with MinGW (filesystem error)

Mr-Precise, 2024-03-02:
### What happened?

Change https://github.com/gnuradio/gnuradio/commit/5173f01788a7e3c95e07286dbfca22e8b75d5273 caused an error when compiling with MinGW gcc.
Before this everything was fine.

### System Information

Host OS: Linux Ubuntu based 20.04 amd64
Compiler: GCC 11.4, mingw-w64 11.0.1 (custom toolchain)
GR Installation Method: Cross-compiling

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-668-gc2d1b5c5

### Steps to Reproduce the Problem

In `gr-fft/lib/fft.cc` I replaced the code blocks wrapped in `_MSC_VER` with `_WIN32`, and the build was successful.
But when trying to use Gnuradio it results in an error:
`filesystem error: cannot create directory: File exists [C:]`

### Relevant log output

```shell
terminate reached from thread id: 124Got std::system_error
filesystem error: cannot create directory: File exists [C:]
```

marcusmueller, 2024-03-02:
hey @Mr-Precise you said you had an error compiling. We need to fix that. The `_MSC_VER` check is there because on MSVC, the arguments to file-handling functions are *not* `char` strings, and I think on MinGW, they should be; maybe I'm mistaken, so we should fix that first. Whether or not what you're seeing is an actual bug or a result of your fix depends on that!

Mr-Precise, 2024-03-02:
My MinGW toolchain is a highly modified [MXE](https://github.com/mxe/mxe)
<details>
  <summary>Build log without "fixes": </summary>
[ 38%] Building CXX object gr-fft/lib/CMakeFiles/gnuradio-fft.dir/fft.cc.obj
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc: In function 'void gr::fft::wisdom_lock_init()':
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:78:36: error: cannot convert 'const value_type*' {aka 'const wchar_t*'} to 'const char*'
   78 |         open(wisdom_lock_file.c_str(), O_WRONLY | O_CREAT | O_NOCTTY | O_NONBLOCK, 0666);
      |              ~~~~~~~~~~~~~~~~~~~~~~^~
      |                                    |
      |                                    const value_type* {aka const wchar_t*}
In file included from /media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/fcntl.h:8,
                 from /media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:29:
/media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/io.h:340:32: note:   initializing argument 1 of 'int open(const char*, int, ...)'
  340 |   int __cdecl open(const char *_Filename,int _OpenFlag,...)  __MINGW_ATTRIB_DEPRECATED_MSVC2005;
      |                    ~~~~~~~~~~~~^~~~~~~~~
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc: In function 'void gr::fft::import_wisdom()':
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:109:39: error: cannot convert 'const value_type*' {aka 'const wchar_t*'} to 'const char*'
  109 |     FILE* fp = fopen(wisdom_path.c_str(), "r");
      |                      ~~~~~~~~~~~~~~~~~^~
      |                                       |
      |                                       const value_type* {aka const wchar_t*}
In file included from /media/buildsystem/MinGW-M/usr/lib/gcc/x86_64-w64-mingw32.shared/11.4.0/include/c++/cstdio:42,
                 from /media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/spdlog/fmt/bundled/core.h:12,
                 from /media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/spdlog/fmt/fmt.h:27,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/io_signature.h:19,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/basic_block.h:15,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/hier_block2.h:15,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/include/gnuradio/top_block.h:15,
                 from /media/buildsystem/gnuradio-main/build/gr-fft/lib/CMakeFiles/gnuradio-fft.dir/cmake_pch.hxx:5,
                 from <command-line>:
/media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/stdio.h:639:49: note:   initializing argument 1 of 'FILE* fopen(const char*, const char*)'
  639 |   FILE *__cdecl fopen(const char * __restrict__ _Filename,const char * __restrict__ _Mode) __MINGW_ATTRIB_DEPRECATED_SEC_WARN;
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc: In function 'void gr::fft::config_threading(int)':
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:123:16: warning: unused variable 'fftw_threads_inited' [-Wunused-variable]
  123 |     static int fftw_threads_inited = 0;
      |                ^~~~~~~~~~~~~~~~~~~
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc: In function 'void gr::fft::export_wisdom()':
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:141:39: error: cannot convert 'const value_type*' {aka 'const wchar_t*'} to 'const char*'
  141 |     FILE* fp = fopen(wisdom_path.c_str(), "w");
      |                      ~~~~~~~~~~~~~~~~~^~
      |                                       |
      |                                       const value_type* {aka const wchar_t*}
In file included from /media/buildsystem/MinGW-M/usr/lib/gcc/x86_64-w64-mingw32.shared/11.4.0/include/c++/cstdio:42,
                 from /media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/spdlog/fmt/bundled/core.h:12,
                 from /media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/spdlog/fmt/fmt.h:27,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/io_signature.h:19,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/basic_block.h:15,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/lib/../include/gnuradio/hier_block2.h:15,
                 from /media/buildsystem/gnuradio-main/gnuradio-runtime/include/gnuradio/top_block.h:15,
                 from /media/buildsystem/gnuradio-main/build/gr-fft/lib/CMakeFiles/gnuradio-fft.dir/cmake_pch.hxx:5,
                 from <command-line>:
/media/buildsystem/MinGW-M/usr/x86_64-w64-mingw32.shared/include/stdio.h:639:49: note:   initializing argument 1 of 'FILE* fopen(const char*, const char*)'
  639 |   FILE *__cdecl fopen(const char * __restrict__ _Filename,const char * __restrict__ _Mode) __MINGW_ATTRIB_DEPRECATED_SEC_WARN;
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc: At global scope:
/media/buildsystem/gnuradio-main/gr-fft/lib/fft.cc:123:16: warning: 'fftw_threads_inited' defined but not used [-Wunused-variable]
  123 |     static int fftw_threads_inited = 0;
      |                ^~~~~~~~~~~~~~~~~~~
make[2]: *** [gr-fft/lib/CMakeFiles/gnuradio-fft.dir/build.make:95: gr-fft/lib/CMakeFiles/gnuradio-fft.dir/fft.cc.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:2001: gr-fft/lib/CMakeFiles/gnuradio-fft.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
</details>

Mr-Precise, 2024-03-02:
Here is a log with the same problem when building Gruradio main branch for gqrx:
https://github.com/Mr-Precise/gqrx-exp/actions/runs/8122920687/job/22202829362

marcusmueller, 2024-03-02:
Interesting! Thanks for the report, helps a lot. Wouldn't have though minGW sufferst the wchar problems! OK, that should be easy to fix.

For future CI reference: https://github.com/Mr-Precise/gqrx-exp/blob/764f11ae1b2b7b09bce678c4867e939465097d60/.github/workflows/ci.yml#L123-L139 is the way you build this, right? By the way, if you can dedicate some time to keep that running, I don't think we'd be averse to adding that minGW builder to our CI infrastructure. But we'd have to rely on you to add the necessary workflow entries in our gnuradio/.github.

Regarding the runtime error:  Yeah. That would qualify as MinGW bug, I think, but the essence is that `C:\Windows\Test` gets split into `C:`, `\`, `Windows`, `Test`. Of which `C:` doesn't exist, `C:\` does exist, `C:\Windows` does exist, `C:\Windows\Test` doesn't exist.

https://gcc.godbolt.org/z/f55bzGrh9

I think a sane way forward here is:

1. Upstream your `_WIN32` fix
2. check for mingw && the path component being the first one &&  being 2 characters long && the second character being `:`; skip the existence check in that case

@Mr-Precise would you agree?

argilo, 2024-03-14:
I ran into the build failure in Gqrx CI as well: https://github.com/gqrx-sdr/gqrx/actions/runs/8284831347/job/22671242447

I guess I'll switch it to build 3.10.9.2 for now.

marcusmueller, 2024-03-14:
yep, this seems to be a MinGW bug in that they split paths incorrectly.

argilo, 2024-03-15:
> yep, this seems to be a MinGW bug in that they split paths incorrectly.

I read over the documentation for `std::filename::path` and I'm not sure it's necessarily a bug. A path can begin with an optional root-name (e.g. `C:`) and root-directory (e.g. `\`), and those parts are included during iteration.

Mr-Precise, 2024-03-15:
#7195 I confirm successful build in MinGW and functionality in gqrx

PR #7222 (2024-03-26): runtime and fft: use create_directories() instead of iterating

## Description
Iterating while creating subdirs in a path was causing problems for MinGW, an is not necessary. Replace with `create_directories()`.

Also use the same fopen() parameters with or without _MSC_VER.

## Related Issue
Fixes #7156

## Which blocks/areas does this affect?
- runtime vmcircbuf config
- fft

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have sign [... cut]
closed 2024-03-26 by willcode