gnuradio/gnuradio #7157: qtgui: remove num_inputs from GUI Sink yml file

willcode, 2024-03-03:
### What happened?

The yml for this block sets multiplicity to `num_inputs`, even though there is no `num_inputs` parameter. The sink does not allow multiple inputs.

### System Information

Source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Place GUI Sink in a flowgraph. The block will cause and error after opening the flowgraph and performing an action such as selecting any block.

Found using the Qt version of GRC, which triggers a check. The Gtk version of GRC does not show this error.

Solution: remove setting of multiplicity from `gr-qtgui/grc/qtgui_sink_x.block.yml` and test

lukazlim, 2024-03-05:
Hi, I can try to work on this issue, may I be assigned?

willcode, 2024-03-05:
Sure. We don't usually assign issues formally - usually people just submit at PR.

Submit the change against the `main` branch, after testing and signing your commit. If you are able (know how to set up multiple builds on your system or switch between versions), also test the change using the `feature-grc-qt` branch, which seems to be more sensitive to the error.

lukazlim, 2024-03-05:
Alright, thanks! By the way, may I have an example screenshot/flowchart that will run into the error for reproducibility?

willcode, 2024-03-05:
Any flowgraph with a Qt GUI Sink, run with `grc --qt` on the `feature-grc-qt` branch. If you want to do the simple fix first, and verify that a flowgraph with a Qt GUI Sink runs under the default GRC, I can try it out with `--qt`. This is more to get you going with a first contribution.

closed 2024-03-07 by willcode