gnuradio/gnuradio #7196: LinuxInstall Wiki Page incorrectly states that latest gnuradio releases are available via PPA

daniel-ayers, 2024-03-15:
### What happened?

The LinuxInstall wiki page at https://wiki.gnuradio.org/index.php?title=LinuxInstall states "For Ubuntu, the latest builds (both released and pulled from master branch) are maintained as PPAs".

That is not correct, for the latest release of Ubuntu (23.10) at least.  The PPA appears to offer packages only for LTS releases of Ubuntu.  If that is the policy of the PPA a note to that effect should be added to the wiki.

In this instance the wiki provides poor advice as the 23.10 system packages are 3.10.7 and the PPA has none.

### System Information

OS: Ubuntu 23.10 (mantic)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

```
# sudo add-apt-repository ppa:gnuradio/gnuradio-releases
Repository: 'Types: deb
URIs: https://ppa.launchpadcontent.net/gnuradio/gnuradio-releases/ubuntu/
Suites: mantic
Components: main
'
Description:
PPA for GNU Radio Tagged Releases
More info: https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Hit:1 http://nz.archive.ubuntu.com/ubuntu mantic InRelease
Hit:2 http://nz.archive.ubuntu.com/ubuntu mantic-updates InRelease                                                                                         
Hit:3 http://nz.archive.ubuntu.com/ubuntu mantic-backports InRelease                                                                                       
Hit:5 https://updates.signal.org/desktop/apt xenial InRelease                                                                                              
Hit:4 https://packages.microsoft.com/repos/code stable InRelease                                                                                           
Hit:6 http://dl.google.com/linux/earth/deb stable InRelease                                                                                                
Hit:7 https://packages.mozilla.org/apt mozilla InRelease                                                                                                   
Hit:8 http://security.ubuntu.com/ubuntu mantic-security InRelease                                                                      
Ign:9 https://ppa.launchpadcontent.net/gnuradio/gnuradio-releases/ubuntu mantic InRelease
Err:10 https://ppa.launchpadcontent.net/gnuradio/gnuradio-releases/ubuntu mantic Release
  404  Not Found [IP: 185.125.190.80 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/gnuradio/gnuradio-releases/ubuntu mantic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

sb424dat, 2024-03-19:
_The PPA appears to offer packages only for LTS releases of Ubuntu._ 

I can add a note to the wiki that LTS releases are supported. Support for non LTS versions is at the risk of the user to support and figure out.

marcusmueller, 2024-04-14:
@daniel-ayers hey, I've added that to the wiki text; would you mind checking out the changed text? Please please reopen this issue if it does not solve the problem (or even better: fix the wiki content!)

The good news is that you get GNU Radio 3.10.7.0 on Ubuntu 23.10 already, which is relatively recent. 

If you want the 3.10.9.2 release (which will be, as of current beta, be in Ubuntu 24.04LTS), you can build that package yourself; since I'm not sure you want that, not going into details here (in essence, `dget` the .dsc from the ubuntu 24.04 release page; remove the versioned dependency on dpkg-dev; mk-build-deps thatfile.dsc; install the resulting .deb using apt-get to get the build dependencies; dpkg-buildpackage -b thatfile.dsc; install the resulting gnuradio*.deb files).

closed 2024-04-14 by marcusmueller