gnuradio/gnuradio #7200: Block rotation in GRC QT do not restrict rotation to [0,359] causing GRC GTK to break when opening

jason-merlo, 2024-03-16:
### What happened?

When using GRC QT, block rotations are integrated, meaning if the block is rotated greater than 270 (360 or above) it will cause GRC GRK to fail to render properly (and even cause the entire GRC GTK UI to start glitching outside of the canvas).

Expected rotation in GRC GTK:
[image]

After rotating to -180 in GRC QT then opening again in GRC GTK:
[image]

### System Information

OS: macOS
GR Installation Method: Source (inside Conda environment)

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-681-gc1d98007

### Steps to Reproduce the Problem

1. Create/Open a flow graph in GRC QT
2. Rotate a block to a total _integrated_ rotation outside [0,90,180,270] (left causes a negative number)
3. Save the flow graph
4. Open in GRC GTK

### Relevant log output

```shell
--- The following is repeated MANY times when opening in GRC GTK ---

Traceback (most recent call last):
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/Application.py", line 788, in _handle_action
    flow_graph_update()
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/Application.py", line 112, in flow_graph_update
    fg.update()
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 203, in update
    self.create_shapes()
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 517, in create_shapes
    element.create_shapes()
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/canvas/block.py", line 135, in create_shapes
    self.bounds_from_area(self._area)
  File "/Users/jason/miniforge3/envs/grdev/lib/python3.12/site-packages/gnuradio/grc/gui/canvas/drawable.py", line 101, in bounds_from_area
    x1, y1, w, h = area
    ^^^^^^^^^^^^
ValueError: not enough values to unpack (expected 4, got 0)
```

willcode, 2024-03-16:
Thanks for the report. Let us know if the simple fix in 7201 works for you.

jason-merlo, 2024-03-17:
That seems to have solve the problem.  Rotating many times in QT and re-opening in GTK works as expected. Thanks!

PR #7201 (2024-03-17): grc-qt: Keep block rotation in [0.0, 360)
Use modulo 360 for block rotation in Qt GRC

Fixes #7200
closed 2024-03-17 by willcode