gnuradio/gnuradio #7208: Xlating FIR Filter NCO does not match the Signal Source implementation

jmfriedt, 2024-03-22:
### What happened?

The following flowchart illustrates how the Xlating FIR NCO local oscillator phase is drifting with respect to the Signal Source. Low-pass filtering the Signal Source output does not impact phase over time (the FIR is not drifting) but the Xlating FIR, whether low-pass filtered or not, is drifting. This drift is frequency proportional.

### System Information

OS: Debian GNU/Linux SID
GR Installation Method: apt install gnuradio-dev
Source: deb http://ftp.fr.debian.org/debian/ sid main contrib non-free

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

See attached:
[image]

willcode, 2024-03-22:
Is this related to #6800?

argilo, 2024-10-01:
> Is this related to https://github.com/gnuradio/gnuradio/issues/6800?

Yes, I believe the root cause is the same. So I'll close this in favour of #6800.

closed 2024-10-01 by argilo