gnuradio/gnuradio #7214: Perf Counter track WORK_CALLED_PRODUCE (and WORK_DONE) as negative produced items

Notou, 2024-03-25:
### What happened?

As mentioned in #7213, the perf counter stat for produced items only uses the work function return value.
And it uses it directly, as an int:
https://github.com/gnuradio/gnuradio/blob/3349d483a1f7e5db0eea042815d237810373857b/gnuradio-runtime/lib/block_executor.cc#L622-L627

So, in cases where the work function uses either WORK_CALLED_PRODUCE, or WORK_DONE, the perf counter counts that as -2 (resp -1) produced items.

### System Information

OS: Linux Mint 21.3
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Make any c++ block, (sync, or not) (It seems gr.WORK_CALLED_PRODUCE is not accepted as return value in Python...)
And have the return value to WORK_CALLED_PRODUCE, setting explicitly produce() and consume() with the right values.

And insert it in whatever flowgraph.
Running gr-ctrlport-monitor, you can see nproduced, avg_nproduced, and var_nproduced at -2 for that block.
[image]

Notou, 2024-03-25:
And more that WORK_CALLED_PRODUCE or WORK_DONE, any negative return value will be taken as face value by the perf counter.

calebbarzee, 2025-01-27:
@Notou I came across this issue looking for additional information on Performance Counters. (BTW, seems like an important bug to fix)

I was wondering if you know why performance counters aren't enabled by default on binary builds that are distributed of GNU Radio? It seems like most hardware would optimize away any performance hit of having these enabled since there is a runtime flag as well.

marcusmueller, 2025-01-27:
hi @calebbarzee! Let's not hijack the bug tracker for questions! I'll love to explain more on the [mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) or in [chat](https://app.element.io/#/room/#gnuradio-space:gnuradio.org), but to give you a quick answer: Perf counters have high performance impact, because they query system clocks, and that can (and usually does) take quite some time, compared to handling a few hundred samples.