gnuradio/gnuradio #7215: work_return_t retrun values not supported in Python

Notou, 2024-03-25:
### What happened?

Returning a work_return_t value (so either gr.WORK_CALLED_PRODUCE or gr.WORK_DONE) at the end of a work function in python leads to a TypeError.

### System Information

OS: Linux Mint 21.3
GR Installation Method: Conda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2

### Steps to Reproduce the Problem

Make an embedded Python block.
Replace the default return value from 
```Python3
return len(output_items[0])
```
With a work_return_t return value. Potentially also explicitly calling consume and produce to replace the return value.
```Python3
self.consume(0, len(output_items[0]))
self.produce(0, len(output_items[0]))
return gr.WORK_CALLED_PRODUCE
```
Insert that block in a flowgraph.
Notice the error message in the console,.

### Relevant log output

```shell
thread_body_wrapper :error: ERROR thread[thread-per-block[1]: <block Embedded Python Block(15)>]: TypeError: '>' not supported between instances of 'gnuradio.gr.gr_python.work_return_t' and 'int'

At:
  /home/cyrille/conda/envs/radioconda/lib/python3.11/site-packages/gnuradio/gr/gateway.py(303): consume_items
  /home/cyrille/conda/envs/radioconda/lib/python3.11/site-packages/gnuradio/gr/gateway.py(200): handle_general_work
```

willcode, 2024-03-25:
Also see #6723

Notou, 2024-03-25:
Ah, yes, good point.
I looked for issues mentioning WORK_CALLED_PRODUCE, but not for WORK_DONE :)