gnuradio/gnuradio #7217: grc cant resolve "@FSM_FILE_INSTALL_DIR@"

ZakyHermawan, 2024-03-26:
### What happened?

In some examples on gr-trellis, variable with id prefix has valule "@FSM_FILE_INSTALL_DIR@".
If a block referencing this prefix variable, the value of this variable will not be resolved (or expanded) as a path, but still remains as string, here's some proof

The value of trellis encoder block is `@FSM_FILE_INSTALL_DIR@awgn102_4.fsm`

[image]

The value of fsm_o is `trellis.fsm(prefix+"/awgn1o2_4.fsm")` which is not a valid file path

[image]

i think this is because FSM_FILE_INSTALL_DIR is defined as a normal variable in `gr-trellis/examples/grc/CMakeLists.txt`

just for additional info,
After changing some missing blocks and hardcoded the value of fsm_o to point to a valid `awgn1o2_4.fsm` path, i'm be able to generate code for turbo_equalization example. And i found that the prefix is also not getting resolved in code generation process, here's the snippet of generated code:

```
class turbo_equalization(gr.top_block, Qt.QWidget):

    def __init__(self):
        ...
        ##################################################
        # Variables
        ##################################################
        self.modulation = modulation = fu.pam4
        self.fsm_o = fsm_o = trellis.fsm("HARD_CODED_PATH_TO_awgn1o2_4.fsm")
        self.channel = channel = fu.c_channel
        self.tot_mod = tot_mod = fu.make_isi_lookup(modulation,channel,False)
        self.bpsym = bpsym = int(round(math.log(fsm_o.I())/math.log(2)))
        self.block_b = block_b = bpsym*1000
        self.EsN0_dB = EsN0_dB = 2
        self.Es = Es = numpy.mean((numpy.square(numpy.abs(tot_mod[1]))))
        self.prefix = prefix = "@FSM_FILE_INSTALL_DIR@"
        self.noisevar = noisevar = 10**(-EsN0_dB/10.0)  * Es   /2.0
        self.interleaver = interleaver = trellis.interleaver(block_b//bpsym,666)
        self.fsm_i = fsm_i = trellis.fsm(len(modulation[1]),len(channel))
        self.R = R = 100e3
        ...
```

### System Information

OS: (Ubuntu 22.04)
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

open any example from gr-trellis (except viterbi_equalization.grc) in grc, and hover to value of variable that is referencing prefix variable

willcode, 2024-03-26:
The expansion of `@FSM_FILE_INSTALL_DIR@` is only meant to work at build/install time so the built-in examples work. Are you trying to use this expansion manually in GRC?

ZakyHermawan, 2024-03-26:
@willcode yes, thats what i'm trying to do, 

btw, to test the examples, is using `$ ctest` is enough ? 
If so, maybe I made a mistake by assuming this was a bug, as the test results from the `ctest` command showed 100% test passed.

You may close this issue if it is indeed not a bug.

willcode, 2024-03-26:
Not a bug. The macro is meant for use during build/install, not at run time.

closed 2024-03-26 by willcode