gnuradio/gnuradio #7227: Filter Design Tool produces error: Module 'scipy' has no attribute 'complex128'

anastas, 2024-03-28:
### What happened?

On Windows 10, the latest gnuradio 3.10.9.2  (installed through radioconda) produces the following 
error when you press the "Design" button.

"Module 'scipy' has no attribute 'complex128'"

I verified that in the installed python 3.11.7,
scipy does not have such a type "complex128".
I also verified that numpy does have such type.

### System Information

OS:Windows 10
GR Installation Method: radioconda

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.9.2 (latest on radioconda)

### Steps to Reproduce the Problem

run gnuradio design tool and press design button

### Relevant log output

```shell
Traceback (most recent call last):
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\scipy\__init__.py", line 137, in __getattr__
    return globals()[name]
           ~~~~~~~~~^^^^^^
KeyError: 'complex128'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\gnuradio\filter\filter_design.py", line 833, in design
    self.design_fir(ftype, fs, gain, winstr)
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\gnuradio\filter\filter_design.py", line 877, in design_fir
    self.draw_plots(taps, params)
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\gnuradio\filter\filter_design.py", line 2329, in draw_plots
    self.update_time_curves()
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\gnuradio\filter\filter_design.py", line 1064, in update_time_curves
    if(type(self.taps[0]) == scipy.complex128):
                             ^^^^^^^^^^^^^^^^
  File "C:\Users\anastas\AppData\Local\radioconda\Lib\site-packages\scipy\__init__.py", line 139, in __getattr__
    raise AttributeError(
AttributeError: Module 'scipy' has no attribute 'complex128'
```

willcode, 2024-03-28:
We should be using non-platform-specific types like `numpy.csingle` and `nump.cdouble`.

See https://numpy.org/doc/stable/reference/arrays.scalars.html

anastas, 2024-03-28:
I traced the error to the different versions of scipy:

On my Linux box I run python3 with scipy 1.8.0 which actually has this type: scipy.complex128

On my Windows10 machine I run python3 with scipy 1.12.0 which DOES NOT have this type.

anastas, 2024-03-28:
The fix is very easy for now:
There are a few places in filter_design.py file where scipy.complex128 appears and needs to be changed to np.complex128.
At least this way all appearances will be np.complex128 instead of mixed scipy/np.complex128.

willcode, 2024-03-28:
If `np.complex128` works for you, that's easier than changing the types. A future goal would still be to use the canonical types.

willcode, 2024-03-28:
Actually, since there are so few occurances of `complex128`, it would be better to just change it to `cdouble`. It get's more complicated with `complex64` since that's also used as a string in places.

willcode, 2024-03-29:
@anastas would you be able to try out the fix in PR 7320?

PR #7230 (2024-03-29): Replace scipy/numpy.complex128 with cdouble
## Description
Use the canonical cdouble instead of the platform-depended complex128 in the filter designer and pmt-to-python code.

The same should be done for complex64 (csingle). This is used more widely in the code, and is also used as a string, so we can address it separately.

## Related Issue
Fixes #7227 

## Which blocks/areas does this affect?
Filter designer tool. Also pmt-to-python.

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits t [... cut]
closed 2024-03-29 by willcode