gnuradio/gnuradio #7242: gr_modtool template choose wrong MANIEST file to install

cxp10, 2024-04-02:
### What happened?

I was following [Creating Python OOT with gr-modtool tutorial](https://wiki.gnuradio.org/index.php?title=Creating_Python_OOT_with_gr-modtool)
This error occurs during `sudo make install` command: 
`CMake Error at cmake_install.cmake:50 (file):`
`  file INSTALL cannot find`
`"/mydirectory/gr-OOT_module/MANIFEST.md": No
  such file or directory.`

Then I  found:
According to [gnuradio/gr-utils/modtool/templates/gr-newmod](https://github.com/gnuradio/gnuradio/tree/3aae3dadb0e0d638f4e5b24d767b4d612cfa63ce/gr-utils/modtool/templates/gr-newmod), gr_modtool generates MANIFEST.yml for a new module.
However, when installing the module, [gnuradio/gr-utils/modtool/templates/gr-newmod/CMakeList.txt](https://github.com/gnuradio/gnuradio/blob/3aae3dadb0e0d638f4e5b24d767b4d612cfa63ce/gr-utils/modtool/templates/gr-newmod/CMakeLists.txt) wants MANIFEST.md instead. `

I created MANIFEST.md manually to fix the error. I expect the MANIFEST file to be managed automatically.

### System Information

OS: Ubuntu 22
GR Installation Method:  Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.9.2-60-gb0766390

### Steps to Reproduce the Problem

create a new module with modtool:
`gr_modtool newmod`
`ModuleName`
`cd gr-ModuleName`
compile and install the module
`mkdir build`
`cd build`
`cmake ..`
`make`
`sudo make install`

willcode, 2024-04-02:
@haakov #7151 did not update `gr-utils/modtool/templates/gr-newmod/CMakeLists.txt`. Could you take a look?

willcode, 2024-04-02:
@haakov simple fix - I've got it.

willcode, 2024-04-02:
@cxp10 thanks for finding this. I'll get the fix into the upcoming release so we don't have a regression.

PR #7245 (2024-04-02): modtool: update CMakeLists.txt template to use new yml vs md format
## Description
The manifest format generated by modtools has changed to YAML but the CMakeLists.txt file was still looking for a md file.

## Related Issue
Fixes #7242 

## Which blocks/areas does this affect?
modtool

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTI [... cut]
closed 2024-04-02 by willcode