gnuradio/gnuradio #7243: gr_modtool failed to install python source file for OOT blocks

cxp10, 2024-04-02:
### What happened?

I was following [Creating Python OOT with gr-modtool tutorial](https://wiki.gnuradio.org/index.php?title=Creating_Python_OOT_with_gr-modtool)
When executing a flowgraph with my OOT block, I encounterd
`Traceback (most recent call last):`
`  File "/gnu_flowgraph/oot_test.py", line 13, in <module>`
`    from gnuradio import OOT_module`
`  File "/usr/local/lib/python3.10/dist-packages/gnuradio/OOT_module/__init__.py", line 23, in <module>`
`    from .python_block import python_block`
`ModuleNotFoundError: No module named 'gnuradio.OOT_module.python_block'`

Then I found my python source file is not installed in dist-packages.
`ls /usr/local/lib/python3.10/dist-packages/gnuradio/OOT_module/`
`__init__.py`

Then I found the `gr_modtool add` command did not change  `/gr-OOT_module/python/OOT_module/CMakeLists.txt`

Probably because [CMakeLists.txt](https://github.com/gnuradio/gnuradio/blob/3aae3dadb0e0d638f4e5b24d767b4d612cfa63ce/gr-utils/modtool/templates/gr-newmod/python/howto/CMakeLists.txt) uses gr_python_install, while [the add command](https://github.com/gnuradio/gnuradio/blob/3aae3dadb0e0d638f4e5b24d767b4d612cfa63ce/gr-utils/modtool/core/add.py)
uses GR_PYTHON_INSTALL, then something went wrong in Regular Expression.

 

### System Information

OS: Ubuntu 22
GR Installation Method:  Source

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.9.2-60-gb0766390

### Steps to Reproduce the Problem

create a new module with modtool:
```
gr_modtool newmod
ModuleName
cd gr-ModuleName
```
The `python/ModuleName/CMakeLists.txt` is generated
add one python block
```
gr_mod add
BlockName
general
python
copyright_holder

n
```
The `python/ModuleName/CMakeLists.txt` is unchanged
For further reproduction :
Add MANIFEST.md[#7242](https://github.com/gnuradio/gnuradio/issues/7242#issue-2219939369) 
Edit parameters in grc/ModuleName_BlockName.yml
Then compile and install the block
```
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```

willcode, 2024-04-02:
CMake function invokation is case-insensitive, even though it's bad style to mix and match and all should be lower case, so that's not it.

Did this work previously? It looks like `_run_python()` is calling `CMakeFileEditor` but then nothing changes.

cxp10, 2024-04-02:
> Reproduced.
> 
> CMake function invokation is case-insensitive, even though it's bad style to mix and match and all should be lower case, so that's not it.
> 
> Did this work previously? It looks like `_run_python()` is calling `CMakeFileEditor` but then nothing changes.

`_run_python()` pass upper case letters to  `CMakeFileEditor`
```
        ed.append_value('GR_PYTHON_INSTALL', fname_py,
                        to_ignore_end='DESTINATION[^()]+')
```

then  `CMakeFileEditor` edit the cmake content with Regular Expression.
```
    def append_value(self, entry, value, to_ignore_start='', to_ignore_end=''):
        """ Add a value to an entry. """
        regexp = re.compile(fr'({entry}\({to_ignore_start}[^()]*?)\s*?(\s?{to_ignore_end})\)',
                            re.MULTILINE)
        substi = r'\1' + self.separator + value + r'\2)'
        (self.cfile, nsubs) = regexp.subn(substi, self.cfile, count=1)
        return nsubs
```
My guess is that Regular Expression there failed to match the upper and the lower case, so the string remains the same.

PR #7246 (2024-04-02): modtools: case-insensitive matching for python cmake file add/rm

## Description

Case of, e.g., GR_PYTHON_INSTALL, is not consistent in cmake files. Since these functions are used in limited situations, it is easier and perhaps more backward compatible to loosen the regex than to iron out the use of case in cmake files.

## Related Issue

Fixes #7243

## Which blocks/areas does this affect?

modtool

## Testing Done
closed 2024-04-02 by willcode