gnuradio/gnuradio #7253: Phase jump occurring in the MMSE resampler output

aguahombre, 2024-04-04:
### What happened?

In ```mmse_fir_interpolator_cc::interpolate``` and ```mmse_resampler_cc_impl::general_work```

For a resample rate > 1.0, when ```d_mu``` increments above 2.0, the filter index ```imu``` wraps from NSTEPS to 0, the sample position will skip forward 2 samples, missing out two samples.
For a resample rate < 1.0, when  ```d_mu``` decrements below 1.0, the filter index``` imu``` wraps from 0 to NSTEPS, the sample position will skip backward 2 samples, repeating the last two samples.

These wrap around edge conditions cause an unwanted phase jump in the resampled data.

mmse_resampler_cc_impl.cc
The lines 87 and 104 should be
```int incr = 2 - (int)f;```

### System Information

Source code in the main repo

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Put ```fprintf``` in the code to output the indices and samples.
See the log output for an example wrap around with resample rate < 1.0

### Relevant log output

```shell
index  incr    imu      d_mu    s - f   output           filter taps
2032	1	0	0.0019	0.0014	-251.42	-264.57  00001000
2033	1	0	0.0014	0.0009	-315.82	-10.57   00001000
2034	1	0	0.0009	0.0004	-332.1	223.49   00001000
2035	0	0	0.0004	0.9999	-302.02	421.08   00001000  <-- incr should be 2
2035	1	128	0.9999	0.9994	-332.1	223.49   00010000  <-- two repeated samples here
2036	1	128	0.9994	0.9989	-302.02	421.08   00010000
2037	1	128	0.9989	0.9984	-232.49	570.36   00010000
2038	1	128	0.9984	0.9979	-134.48	665.44   00010000
2039	1	128	0.9979	0.9974	-21.35	706.71
```

willcode, 2024-04-04:
Can you clarify whether this is only a problem on dev-4.0? That branch is inactive (see the README).

willcode, 2024-04-04:
OK. Your writeup says dev-4.0.

aguahombre, 2024-04-04:
The filter taps for imu  = 0 are 
0, 0, 0, 0, 1, 0, 0, 0
and for imu = 128 : 
0, 0, 0, 1, 0, 0, 0, 0
When d_mu_inc < 1.0 , and imu rolls over from 0 to 128, there is a phase change of 1 sample backwards, so the increment needs to be 2 not 1 in this case, to adjust for the phase shift caused by the tap delay of 1.

imu rolls over from 0 to 128, when f  < 1.0, therefore (int)f == 0,  so 2 -(int)f == 2 as required.

2-(int)f where f is
0 => 2
1 => 1
2 => 0

Likewise when d_mu_inc is > 1.0, and imu rolls over from 128 to 0, there is a phase change of 1 sample forwards, and f will be > 2, therefore (int)f will be 2,
but incr needs to be 0; i.e. 2 - (int)f

Hope this helps

marcusmueller, 2024-04-09:
@willcode on it

marcusmueller, 2024-04-09:
Notes as I look at this

- How do we know that in ` out[oo++] = d_resamp.interpolate(&in[ii], static_cast<float>(d_mu));` (L. 83) `ii + d_resamp.ntaps()` is not OOB for `in`?
-

marcusmueller, 2025-02-21:
hey @aguahombre does #7743 solve the issue for you?

daniestevez, 2025-02-25:
I've arrived here through #7743, and I don't think this report is right. If there are jumps in the MMSE resampler, it should be possible to come up with an example that shows those jumps in the output signal.

The bug report says that for a resampling rate > 1.0, sometimes the filter arm wraps from NSTEPS to 0, and then the sample index is incremented by 2. It is claimed that this is a discontinuity. I don't agree that this is a discontinuity.

This MMSE resampler works with the "polyphase" filter defined in https://github.com/gnuradio/gnuradio/blob/main/gr-filter/include/gnuradio/filter/interpolator_taps.h The taps for arm = 0 are  `(0, 0, 0, 0, 1, 0, 0, 0)`, while the taps for arm = NSTEPS are `(0, 0, 0, 1, 0, 0, 0, 0)`. 

Quite fundamentally to understand this issue, the MMSE resampler uses the `fir_filter` class for filtering. This uses the taps in reverse order, as is common in the mathematical definition of an FIR filter: $y_{n + k - 1} = x_{n-k+1} a_{k-1} + x_{n-k+2} a_{k-2} + \cdots + x_{n-1} a_1 + x_n a_0$ (where $x_j$ are the input samples and $a_j$ are the taps). The operation itself is performed with a Volk dot-product, which doesn't do this reverse order, but the [taps are reversed](https://github.com/gnuradio/gnuradio/blob/main/gr-filter/lib/fir_filter.cc#L34) when they are set in the filter.

With this in mind, let's say we have a resampling rate of 1.0001 (close to 1.0 but slightly greater), and, as the report says, the arm wraps from NSTEPS to 0 while the sample index is incremented by 2. Because of the NSTEPS -> 0 wrap, the 1 in the FIR filter has shifted one position to the right. Due to the reversal of the FIR filter taps, the FIR filter is picking one sample earlier compared to what it did before. Since the sample index has been incremented by 2, the combined effect is that we have advanced by one sample. This is what is to be expected for a resampling rate very close to 1, so there is no discontinuity.

Having explained all this, it seems to me that @aguahombre has forgotten to consider that the filter taps are used in reverse order. The way I think about this to avoid mistakes is that the arm index of the MMSE corresponds to a fractional sampling offset between 0.0 samples (for arm = 0) and 1.0 samples (for arm = NSTEPS), with positive offsets meaning later samples in the stream. Therefore, if we have a resampling rate close to 1.0, it is to be expected that when the MMSE offset rolls from near 1.0 to near 0.0, there must be an additional +1 increment in the sample index to compensate for the roll (and this situation happens regularly with resample rate > 1.0 as reported). Likewise, when rolling from near 0.0 to near 1.0, there must be an additional -1 increment in the sample index to compensate for the roll (and this situation happens regularly with resample rate < 1.0 as reported).

closed 2025-03-08 by marcusmueller