gnuradio/gnuradio #7265: Documentation for Max block is incorrect

mormj, 2024-04-12:
### What happened?

[image]

Documentation on the max block seems to include info from other blocks

### System Information

OS: Ubuntu 22.04
GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Open the properties of the Max block

ZakyHermawan, 2024-04-15:
In _update_docs_page method from class PropsDialog, im trying to print docstrings variable right after `docstrings = self._block.documentation.copy()`, here's the output
[image]

As you can see, from the beginning, there is matrix interleaver docs in the docstrings dict, this docs is taken from matrix_interleavers.py

ZakyHermawan, 2024-04-15:
are there any reason for a block to have docs from yml file and the python file at the same time ? isn't the python file for a block is actually generated from yml file ?

We can quick fix this issue by just return right after yaml file being parsed:
```python
if from_yaml:
  buf.insert(pos, '\n')
  return

```
[image]

ffu, 2025-01-05:
This bug can be traced to function docstring_guess_from_key in module extract_docs.py. The handling of block names with variable types like blocks_max_xx is incorrect. The function uses a regex to search for matching block names:

    pattern = re.compile(
        '^' + init_name.replace('_', '_*').replace('x', r'\w') + r'\w*$')
        
However, for an init_name like "max_xx", the regex becomes r"^ma\w_*\w\w\w*$". The x in "max" is replaced with "\w" and the underscore is optional.  Therefore, any block that begins with "ma" and contains at least four characters become a matching block for max_xx. This includes magphase_to_complex and matrix_interleaver. Eventually, the documentation for the max block also contains docstrings for magphase_to_complex and matrix_interleaver.

A possible solution is to check if the block name ends with _x, _xx, _xxx, or _vxx. If so, then only replace x with "\w" in the suffix.

commit 0867f7fd (2025-01-21): grc: Fix documentation showing wrong block

This commit fixes issues where the documentation tab shows
the wrong block. This was first noticed when the max
block showed documentation for magphase_to_complex and
matrix_interleaver.
fixes #7265
commit 23b2ba9d (2025-01-22): grc: Fix documentation showing wrong block

This commit fixes issues where the documentation tab shows
the wrong block. This was first noticed when the max
block showed documentation for magphase_to_complex and
matrix_interleaver.
fixes #7265
closed 2025-01-21 by willcode