gnuradio/gnuradio #7266: Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X? when running gnuradio-companion

santfirax, 2024-04-15:
### What happened?

I followed https://wiki.gnuradio.org/index.php/CondaInstall to install gnuradio in my ubuntu install with WSL but when trying to run gnuradio-companion I get the following error
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'GObject', version '2.0' not found)

### System Information

Windows using WSL2 ubuntu
GR Installation Method: Conda

### GNU Radio Version

4.0-git (dev-4.0)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Follow https://wiki.gnuradio.org/index.php/CondaInstall for Linux

### Relevant log output

```shell
(gnuradio) smolano@DESKTOP-6V1M217:~$ gnuradio-companion
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'GObject', version '2.0' not found)
(gnuradio) smolano@DESKTOP-6V1M217:~$ gnuradio-companion -X
ImportError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(Typelib file for namespace 'GObject', version '2.0' not found)
(gnuradio) smolano@DESKTOP-6V1M217:~$
```

marcusmueller, 2024-04-15:
If you're running a conda-installed version in WSL2, your host system probably needs to run an X server, or you need to use a current version of WSL that supports WSLg:

Two solutions:
1. install GNU Radio natively on windows using conda (no WSL) 
2. install an X server; see https://wiki.gnuradio.org/index.php/WindowsInstall#WSL_1/2_(before_WSLg)

Generally, there's no real downside to directly installing on Windows, and performance of display would be better, and because our current GNU Radio installer (based on conda) for windows is a quick thing to install (and remove), would you want to try the 1. solution? You'd just need to download and run the "Windows Radioconda Installer" from https://wiki.gnuradio.org/index.php?title=InstallingGR#Quick_Start ; I'm not a Windows user, but I think normal user privileges should suffice for that.

willcode, 2024-04-15:
Closing - please use chat or the mailing list for help (this is not a bug).

closed 2024-04-15 by willcode

vaughanatworld, 2024-04-18:
Seems like a bug to me.
I'm running Debian 11 with X, and installed radioconda and I can see the GObject library and .h file under anaconda3.
I get the exact same error message.